usb-tools/USBProxy-legacy #40: Can't get it work with dummy_hcd

matlo, 2015-03-27:
Hi,
I'm trying to make USBProxy work with dummy_hcd.
I'm using Ubuntu 14.04 which runs a 3.13 linux kernel.
I built the gadgetfs and dummy_hcd modules:
```
apt-get source linux-image-`uname -r`
cd linux-3.13.0/
cp /boot/config-`uname -r` .config
cp /usr/src/linux-headers-`uname -r`/Module.symvers .
make menuconfig
Device Drivers > USB support > USB Gadget Support > Dummy HCD
make prepare
make scripts
make -j 4 M=drivers/usb/gadget
sudo cp drivers/usb/gadget/{gadgetfs,dummy_hcd}.ko /lib/modules/`uname -r`/kernel/drivers/usb/gadget/
sudo depmod -a
```
I loaded the dummy_hcd module:
```
sudo modprobe dummy_hcd
```
And I started usb-mitm:
```
sudo usb-mitm -v 046d -p c07d
Loading plugins from /usr/local/lib/USBProxy/
vendorId=046d
productId=c07d
cleaning up /tmp
removing 0
Made directory /tmp/gadget-YH5jnE for gadget
Printing Config data
	Strings: 4
		DeviceProxy: DeviceProxy_LibUSB
		HostProxy: HostProxy_GadgetFS
		productId: c07d
		vendorId: 046d
	Vectors: 0
Pointer: 0
Device: 12 01 00 02 00 00 00 40 6d 04 7d c0 02 88 01 02 03 01
  Manufacturer: Logitech
  Product:      Gaming Mouse G502
  Serial:       *************
	*Config(1): 09 02 3b 00 02 01 04 a0 96
	   Name: U88.02_B0016            
		Interface(0):
			*Alt(0): 09 04 00 00 01 03 01 02 00
				HID: 09 21 11 01 00 01 22 43 00
				EP(81): 07 05 81 03 08 00 01
		Interface(1):
			*Alt(0): 09 04 01 00 01 03 00 00 00
				HID: 09 21 11 01 00 01 22 97 00
				EP(82): 07 05 82 03 14 00 01
searching in [/tmp/gadget-YH5jnE]
Starting setup reader thread (2776) for EP00.
Starting setup writer thread (2777) for EP00.
Fail on open EP81 2 No such file or directory
Starting reader thread (2778) for EP81.
Starting writer thread (2781) for EP82.
Starting reader thread (2780) for EP82.
Starting writer thread (2779) for EP81.
Received SIGHUP, restarting relaying...
Finished writer thread (2781) for EP82.
Finished setup writer thread (2777) for EP00.
Finished writer thread (2779) for EP81.
Finished reader thread (2778) for EP81.
Finished reader thread (2780) for EP82.
Finished setup reader thread (2776) for EP00.
Error unmounting gadgetfs from [/tmp/gadget-YH5jnE].
Error removing directory [/tmp/gadget-YH5jnE].
Device: 12 01 00 02 00 00 00 40 6d 04 7d c0 02 88 01 02 03 01
  Manufacturer: Logitech
  Product:      Gaming Mouse G502
  Serial:       *************
	*Config(1): 09 02 3b 00 02 01 04 a0 96
	   Name: U88.02_B0016            
		Interface(0):
			*Alt(0): 09 04 00 00 01 03 01 02 00
				HID: 09 21 11 01 00 01 22 43 00
				EP(81): 07 05 81 03 08 00 01
		Interface(1):
			*Alt(0): 09 04 01 00 01 03 00 00 00
				HID: 09 21 11 01 00 01 22 97 00
				EP(82): 07 05 82 03 14 00 01
cleaning up /tmp
removing 1
Made directory /tmp/gadget-tbXzNh for gadget
searching in [/tmp/gadget-tbXzNh]
Fail on open 16 Device or resource busy
GadgetFS not connected.
done
```
Here is the output of dmesg:
```
dmesg
[   44.763401] dummy_hcd dummy_hcd.0: USB Host+Gadget Emulator, driver 02 May 2005
[   44.763405] dummy_hcd dummy_hcd.0: Dummy host controller
[   44.763408] dummy_hcd dummy_hcd.0: new USB bus registered, assigned bus number 5
[   44.763438] usb usb5: New USB device found, idVendor=1d6b, idProduct=0002
[   44.763440] usb usb5: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[   44.763441] usb usb5: Product: Dummy host controller
[   44.763443] usb usb5: Manufacturer: Linux 3.13.0-48-generic dummy_hcd
[   44.763444] usb usb5: SerialNumber: dummy_hcd.0
[   44.764770] hub 5-0:1.0: USB hub found
[   44.764778] hub 5-0:1.0: 1 port detected
[   56.359823] gadgetfs: USB Gadget filesystem, version 24 Aug 2004
[   56.426988] gadgetfs: bound to dummy_udc driver
[   56.641830] usb 5-1: new high-speed USB device number 2 using dummy_hcd
[   56.645730] gadgetfs: connected
[   56.645754] gadgetfs: disconnected
[   56.781752] gadgetfs: connected
[   56.821777] usb 5-1: New USB device found, idVendor=046d, idProduct=c07d
[   56.821782] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   56.821785] usb 5-1: Product: Gaming Mouse G502
[   56.821788] usb 5-1: Manufacturer: Logitech
[   56.821790] usb 5-1: SerialNumber: ***********
[   56.825765] gadgetfs: configuration #1
[   56.862044] input: Logitech Gaming Mouse G502 as /devices/platform/dummy_hcd.0/usb5/5-1/5-1:1.0/input/input17
[   56.862160] hid-generic 0003:046D:C07D.0007: input,hidraw4: USB HID v1.11 Mouse [Logitech Gaming Mouse G502] on usb-dummy_hcd.0-1/input0
[   56.909900] input: Logitech Gaming Mouse G502 as /devices/platform/dummy_hcd.0/usb5/5-1/5-1:1.1/input/input18
[   56.909986] hid-generic 0003:046D:C07D.0008: input,hiddev0,hidraw5: USB HID v1.11 Keyboard [Logitech Gaming Mouse G502] on usb-dummy_hcd.0-1/input1
[   56.933786] dummy_hcd dummy_hcd.0: timer fired with no URBs pending?
[   58.409577] input: Logitech Gaming Mouse G502 as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5:1.0/input/input19
[   58.409932] hid-generic 0003:046D:C07D.0009: input,hidraw6: USB HID v1.11 Mouse [Logitech Gaming Mouse G502] on usb-0000:00:1d.0-1.5/input0
[   58.412403] input: Logitech Gaming Mouse G502 as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5:1.1/input/input20
[   58.412515] hid-generic 0003:046D:C07D.000A: input,hiddev0,hidraw7: USB HID v1.11 Keyboard [Logitech Gaming Mouse G502] on usb-0000:00:1d.0-1.5/input1
[   58.412538] gadgetfs: disconnected
[   58.412700] usb 5-1: USB disconnect, device number 
```
It seems something wrong happens at 56.933786.
Has anyone managed to get USBProxy work with dummy_hcd?
Any suggestion?

dominicgs, 2015-03-27:
I haven't looked at dummy_hcd before, but it looks like a very useful tool to pair with USBProxy.

I think the problem is that we don't yet support the controllers that dummy_hcd emulates, but it should only be a minor change to add support.

Some things to try first:
1) Try dummy_hcd with one of the pre-built gadgets, such as the serial device or mass storage. This just makes sure that it's not a problem with dummy_hcd on your system.
2) Try the following:
```
modprobe dummy_hcd
modprobe gadgetfs
mkdir /tmp/gadget
mount gadgetfs /tmp/gadget
ls /tmp/gadget
```
The filesystem should contain a single file that specifies which type of controller it is emulating.  Then hopefully we can just add that to the list that we check for.

dominicgs, 2015-03-27:
Oh, wait, I misread the output.  We clearly support it as the device appears.  This looks like it's some deeper bug.

matlo, 2015-03-27:
I started usb-mitm with gdb, and stopped the execution at the open_endpoint() function.
The program tries to open /tmp/gadget-ZgqpOH/ep1in, but there's no such file:
```
ll /tmp/gadget-ZgqpOH/
total 0
-rw------- 1 root root 0 mars  27 19:45 dummy_udc
-rw------- 1 root root 0 mars  27 19:45 ep10in-int
-rw------- 1 root root 0 mars  27 19:45 ep10out
-rw------- 1 root root 0 mars  27 19:45 ep11in-bulk
-rw------- 1 root root 0 mars  27 19:45 ep11out
-rw------- 1 root root 0 mars  27 19:45 ep12in
-rw------- 1 root root 0 mars  27 19:45 ep12out-bulk
-rw------- 1 root root 0 mars  27 19:45 ep13in-iso
-rw------- 1 root root 0 mars  27 19:45 ep13out
-rw------- 1 root root 0 mars  27 19:45 ep14in
-rw------- 1 root root 0 mars  27 19:45 ep14out-iso
-rw------- 1 root root 0 mars  27 19:45 ep15in-int
-rw------- 1 root root 0 mars  27 19:45 ep15out
-rw------- 1 root root 0 mars  27 19:45 ep1in-bulk
-rw------- 1 root root 0 mars  27 19:45 ep1out-bulk
-rw------- 1 root root 0 mars  27 19:45 ep2in-bulk
-rw------- 1 root root 0 mars  27 19:45 ep2out-bulk
-rw------- 1 root root 0 mars  27 19:45 ep3in-iso
-rw------- 1 root root 0 mars  27 19:45 ep3out
-rw------- 1 root root 0 mars  27 19:45 ep4in
-rw------- 1 root root 0 mars  27 19:45 ep4out-iso
-rw------- 1 root root 0 mars  27 19:45 ep5in-int
-rw------- 1 root root 0 mars  27 19:45 ep5out
-rw------- 1 root root 0 mars  27 19:45 ep6in-bulk
-rw------- 1 root root 0 mars  27 19:45 ep6out
-rw------- 1 root root 0 mars  27 19:45 ep7in
-rw------- 1 root root 0 mars  27 19:45 ep7out-bulk
-rw------- 1 root root 0 mars  27 19:45 ep8in-iso
-rw------- 1 root root 0 mars  27 19:45 ep8out
-rw------- 1 root root 0 mars  27 19:45 ep9in
-rw------- 1 root root 0 mars  27 19:45 ep9out-iso
```

dominicgs, 2015-03-27:
Right, but there's a file called ```ep1in-bulk``` which we should be opening instead.

This is confusing, that's a very large list of endpoints, most of which don't exist in the descriptors that we wrote to dummy_udc.

I built and ran this locally earlier today and hit a different error, but I think it was probably related.  I was having reads fail from one of the files.

dominicgs, 2015-03-27:
Looks like this line needs to be updated to also check for dummy_udc: https://github.com/dominicgs/USBProxy/blob/master/src/Plugins/Hosts/GadgetFS_helpers.c#L247

matlo, 2015-03-27:
I hardcoded "in-bulk" and got it to work :)
There's some packet buffering, I think my mouse is too fast (bInterval is 1ms).
I also had to comment out the libusb_hotplug_register_callback stuff in DeviceProxy_LibUSB.cpp because usb-mitm fails at unmouting gadgetfs...

dominicgs, 2015-03-27:
Yes, the unmounting thing is strange, I don't see that behaviour with real gadget devices.

There's a bug where the gadgetfs module "used by" count is -1.  This is something we see on the BBB but it uses an unsigned int rather than a signed one (which seems strange).

I'll probably fix the logic to match the ```sw_usb_udc``` device.

matlo, 2015-03-27:
Actually, ep1in-bulk means that the endpoint has a fixed (direction, type).
I think we should probe the available endpoints and map each device endpoint to the first available endpoint of that can have the same direction and type.
Example with dummy_hcd:
endpoint 1, interrupt in => ep4in
endpoint 2, interrupt in => ep5in-int
What do you think about it?

dominicgs, 2015-08-19:
@matlo Did you get any further with this issue?  I've just merged your other changes and would be keen for dummy_hcd support.

matlo, 2015-09-01:
No I didn't. I'm not sure I'll have time to work on it.

straithe, 2021-11-13:
I'm sorry you haven't gotten a response in a while, @matlo. Are you still experiencing this issue?

closed 2021-12-12 by straithe