usb-tools/USBProxy-legacy #41: Code formatting

matlo, 2015-04-03:
I think the source code formatting needs to be reviewed.
I suggest using the  [K&R style](http://en.wikipedia.org/wiki/Indent_style#K.26R_style).
It is one of the built-in code style formatter available in eclipse.
I can format all the source code and submit a pull request if you want.

dominicgs, 2015-04-03:
We're not too far from K&R style already, are we?  I don't really mind which style is used and I agree that it should aim to be consistent.

However, I very much dislike commits that change large parts of the codebase without changing any functionality.  They make it more difficult to find where code came from and which changes are related.

What I'm asking is this:
 * how much of a problem is the current style?
 * how big would the change be?
 * would we be better just asking developers/contributors to adhere to a specific style?

matlo, 2015-04-03:
This can be found in src/Plugins/Devices/DeviceProxy_LibUSB.cpp:
```c++
if (dev_handle) {fprintf(stderr,"LibUSB already connected.\n"); return 0;}
```
What you get after using the K&R style formatter:
```c++
if (dev_handle) {
	fprintf(stderr, "LibUSB already connected.\n");
	return 0;
}
```
Having a condition and multiple statements on the same line is just not readable.
This is just an example. There are actually a lot of things that should be fixed to improve readability:
- spaces between operands
- spaces between arguments
- very long lines
- ...

There also are other things to fix that are not related to code formatting:
- hardcoded values (there are a few of them in DeviceProxy_LibUSB.cpp)
- mixing "std::cerr << ..." and "fprintf(stderr, ..."

jaapcrezee, 2015-04-03:
I do not like K&R style, it just feels old and dusty:

> very long lines

We are no longer living in the year 1980; I really do not mind having lines with 160 chars on it, it will fit anyway on any modern display.

I agree with Dominic we should not make large commits without any functional change, it will make looking at changes a lot complexer in the future using data from the past before the code-style changes.

If you ask me, we should just agree on a certain style and use that from now on.
When things are really unreadable right now, as state above, we should just change /that/.

dominicgs, 2015-04-03:
@matlo You're right, I had forgotten that those if statements bother me too.  I think they qualify as "unreadable right now".  If you send me a pull request for them I will happily accept it.

The same goes for the switching fprintf to std::cerr.

Spaces between operands / arguments just doesn't strike me as a big deal right now, but if you fix them in the course of fixing other things, then that's fine too.

aholler, 2015-04-03:
Also I hate style discussions as there are as many styles as devs and projects and I believe a good dev should be able to read any style, here are my 2¢:

- Multiple statements in one line are making debugging harder because many debuggers are showing only the current line and not the current statement.
- Multiple statements in one line don't work that good with source code versioning systems. And it makes it much harder to see which statement was changed if there are multiple in one line.
- I would change existing code only if it's touched.
- Please don't start with those silly style patches as done by kernel monkeys.

jaapcrezee, 2015-04-03:
Ain't this the gd-truth :)
Amen.

aholler, 2015-04-03:
Sorry, but I've no idea what you mean with gd-truth.

I just hate all the style discussions und patch refusals because of style as done by kernel-people and think they should waste their time with fixing real bugs instead of discussing styles. Style isn't syntax and everyone sees/reads code different because brains are different.

aholler, 2015-04-03:
I also have no problem with having different styles in one project. This helps mapping code to people (you see which code was written by whom). ;)

closed 2015-06-26 by dominicgs