usb-tools/USBProxy-legacy #47: Segmentation fault with Usb-to-ethernet interface

hllhll, 2015-08-25:
Can't get passed segmentation fault after connecting the Beagle to the pc:
Steps to repreduce:

Grab a d-link ethernet to usb converter,
Windows 7 mechine
Stock drivers of the dlink for win64 (On the cd, ver XXX)
USBProxy version - GIT head

Execute:

root@arm:/# usb-mitm -dd
Version ShmooCon_2015-71-gb0f9
Running under kernel 3.12.0-bone8
CP: String DeviceProxy = DeviceProxy_LibUSB
CP: String HostProxy = HostProxy_GadgetFS
Loading plugins from /usr/local/lib/USBProxy/
vendorId=ffffffff
productId=ffffffff
cleaning up /tmp
removing 0
Made directory /tmp/gadget-8zio46 for gadget
Printing Config data
        Strings: 2
                DeviceProxy: DeviceProxy_LibUSB
                HostProxy: HostProxy_GadgetFS
        Vectors: 0
Pointer: 0
Connected to device: 2001:1a02@05 D-Link           - DUB-E100

LibUSB> 80 06 00 01 00 00 12 00
LibUSB< 12 01 00 02 ff ff 00 40 01 20 02 1a 01 00 01 02 03 01
LibUSB> 80 06 00 03 00 00 08 00
LibUSB< 04 03 09 04
LibUSB> 80 06 01 03 09 04 08 00
LibUSB< 22 03 44 00 2d 00 4c 00
LibUSB> 80 06 01 03 09 04 22 00
LibUSB<
        22 03 44 00 2d 00 4c 00 69 00 6e 00 6b 00 20 00 20 00 20 00 20 00 20 00 20 00 20 00 20 00 20 00
        20 00
LibUSB> 80 06 02 03 09 04 08 00
LibUSB< 12 03 44 00 55 00 42 00
LibUSB> 80 06 02 03 09 04 12 00
LibUSB< 12 03 44 00 55 00 42 00 2d 00 45 00 31 00 30 00 30 00
LibUSB> 80 06 03 03 09 04 08 00
LibUSB< 0e 03 36 00 44 00 41 00
LibUSB> 80 06 03 03 09 04 0e 00
LibUSB< 0e 03 36 00 44 00 41 00 44 00 37 00 33 00
LibUSB> 80 06 00 02 00 00 09 00
LibUSB< 09 02 27 00 01 01 04 a0 64
LibUSB> 80 06 00 02 00 00 27 00
LibUSB<
        09 02 27 00 01 01 04 a0 64 09 04 00 00 03 ff ff 00 07 07 05 81 03 08 00 0b 07 05 82 02 00 02 00
        07 05 03 02 00 02 00
LibUSB> 80 06 04 03 09 04 08 00
LibUSB< 04 03 30 00
LibUSB> 80 06 07 03 09 04 08 00
LibUSB< 04 03 30 00
LibUSB> 80 06 00 06 00 00 0a 00
LibUSB< 0a 06 00 02 ff ff 00 08 01 00
LibUSB> 80 06 00 07 00 00 09 00
LibUSB< 09 07 27 00 01 01 04 a0 64
LibUSB> 80 06 00 07 00 00 27 00
LibUSB<
        09 07 27 00 01 01 04 a0 64 09 04 00 00 03 ff ff 00 07 07 05 81 03 08 00 a0 07 05 82 02 40 00 00
        07 05 03 02 40 00 00
LibUSB> 80 06 04 03 09 04 08 00
LibUSB< 04 03 30 00
LibUSB> 80 06 07 03 09 04 08 00
LibUSB< 04 03 30 00
LibUSB> 80 08 00 00 00 00 01 00
LibUSB< 01
Device: 12 01 00 02 ff ff 00 40 01 20 02 1a 01 00 01 02 03 01
  Manufacturer: D-Link
  Product:      DUB-E100
  Serial:       6DAD73
        *Config(1): 09 02 27 00 01 01 04 a0 64
           Name: 0
                Interface(0):
                        *Alt(0): 09 04 00 00 03 ff ff 00 07
                           Name: 0
                                EP(81): 07 05 81 03 08 00 0b
                                EP(82): 07 05 82 02 00 02 00
                                EP(03): 07 05 03 02 00 02 00
HS Qualifier: 0a 06 00 02 ff ff 00 08 01 00
         Config(1): 09 07 27 00 01 01 04 a0 64
           Name: 0
                Interface(0):
                        *Alt(0): 09 04 00 00 03 ff ff 00 07
                           Name: 0
                                EP(81): 07 05 81 03 08 00 a0
                                EP(82): 07 05 82 02 40 00 00
                                EP(03): 07 05 03 02 40 00 00

        00 00 00 00 09 02 27 00 01 01 04 80 64 09 04 00 00 03 ff ff 00 07 07 05 81 03 08 00 a0 07 05 82
        02 40 00 00 07 05 03 02 40 00 00 09 02 27 00 01 01 04 80 64 09 04 00 00 03 ff ff 00 07 07 05 81
        03 08 00 0b 07 05 82 02 00 02 00 07 05 03 02 00 02 00 12 01 00 02 ff ff 00 40 01 20 02 1a 01 00
        01 02 03 01
searching in [/tmp/gadget-8zio46]
Starting setup writer thread (2414) for EP00.
Starting setup reader thread (2413) for EP00.

<CONNECTING BEAGLE TO PC>

gadgetfs: 1 events received
gadgetfs: event 1
gadgetfs: 1 events received
gadgetfs: event 2
gadgetfs: 1 events received
gadgetfs: 1 events received
LibUSB> 80 06 03 03 09 04 ff 00
LibUSB< 0e 03 36 00 44 00 41 00 44 00 37 00 33 00
gadgetfs: 1 events received
GADGET_SETUP
LibUSB> 80 06 00 03 00 00 ff 00
LibUSB< 04 03 09 04
gadgetfs: 1 events received
LibUSB> 80 06 02 03 09 04 ff 00
LibUSB< 12 03 44 00 55 00 42 00 2d 00 45 00 31 00 30 00 30 00
gadgetfs: 1 events 
LibUSB> 00 09 01 00 00 00 00 00
LibUSB<
Opened EP81
Opened EP82
Opened EP03
Sending ACK
Starting reader thread (2432) for EP03.
Starting reader thread (2430) for EP82.
Starting writer thread (2433) for EP03.
Starting writer thread (2431) for EP82.
Starting writer thread (2429) for EP81.
Starting reader thread (2428) for EP81.
gadgetfs: 1 events received
LibUSB> 00 09 00 00 00 00 00 00
LibUSB<
Segmentation fault
root@arm:/#

hllhll, 2015-08-25:
Additional information:

Running it through gdb gives out:

[Similar output]
LibUSB> 00 09 00 00 00 00 00 00
LibUSB<

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xb5ce6460 (LWP 2459)]
0xb6ceaea4 in HostProxy_GadgetFS::setConfig(Configuration*, Configuration*, bool) ()
   from /usr/local/lib/USBProxy/HostProxy_GadgetFS.so
(gdb)

dominicgs, 2015-08-25:
Are you using the provided OS image? If not, did you apply the kernel patch?

Could you paste the output of lsusb -v for the device here?

hllhll, 2015-08-25:
[Wow, that's a fast response]
Yes, I'm using the release image with an updated software from the git.
Also need to note that without the windows driver, it does not crash, untill that Is I install the driver.

I'll be on the win machine soon and I'll be able to give it out but right now I more in need of other issue (Should I Open another ticket?):

On another machine (Linux-like) the device does not crash, however, it says:

LibUSB> c0 11 00 00 00 00 03 00
LibUSB< 15 0c 12
Fail on EP00 write 22 Invalid argument
Transfer error on EP82 (xfertype 2): Operation timed out
Transfer error on EP82 (xfertype 2): Operation timed out
...[to infinity]...

The device is `seen` correctly on the nix machine, but the iterface itslef is not communicative (the os could not set it to `up` state). No problem when connecting directly without mitm.

hllhll, 2015-08-25:
lsusb
```
root@arm:~# lsusb -v

Bus 002 Device 003: ID 2001:1a02 D-Link Corp. DUB-E100 Fast Ethernet Adapter(rev.C1) [ASIX AX88772]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass       255 Vendor Specific Subclass
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x2001 D-Link Corp.
  idProduct          0x1a02 DUB-E100 Fast Ethernet Adapter(rev.C1) [ASIX AX88772]
  bcdDevice            0.01
  iManufacturer           1 D-Link          
  iProduct                2 DUB-E100
  iSerial                 3 6DAD73
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           39
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          4 0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              200mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol      0 
      iInterface              7 0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              11
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
[... 118 lines cut ...]
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0011
    Per-port power switching
    No overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood        5 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x02
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0100 power
Device Status:     0x0001
  Self Powered
root@arm:~# 

```

dominicgs, 2015-08-25:
There's an interrupt endpoint.  I would imagine that may be causing us to fall over, although we should be printing an error message rather than segfaulting.

hllhll, 2015-08-25:
And the timeout issue? any chance I could address that?

dominicgs, 2015-08-25:
I'm not sure, EP82 should work, it's a bulk endpoint which we support for many other devices.

eliocor, 2016-01-10:
I'm getting the same segfault!
Working environment:
Windows 10
BeagleBone Black with OS image: https://github.com/dominicgs/USBProxy/releases/download/2014-03-R1/Debian-USBProxy.img.xz
Source files updated to today: https://github.com/dominicgs/USBProxy
makefile generated with '**cmake -DCMAKE_BUILD_TYPE=Debug ..**' command
gcc version: 4.7.2-5

**lsusb -v** output of the device I'm trying to analyze (Gretag MacBeth EyeOne Pro):
```
Bus 002 Device 013: ID 0971:2000 Gretag-Macbeth AG
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x0971 Gretag-Macbeth AG
  idProduct          0x2000
  bcdDevice            2.00
  iManufacturer           1 GretagMacbeth
  iProduct                2 eye-one
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           46
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           4
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
[... 94 lines cut ...]
Starting setup reader thread (5649) for EP00.
[New Thread 0xb5ce6460 (LWP 5650)]
Starting setup writer thread (5650) for EP00.
gadgetfs: 1 events received
gadgetfs: event 1
gadgetfs: 1 events received
LibUSB> 80 06 00 03 00 00 ff 00
LibUSB< 04 03 09 04
gadgetfs: 1 events received
LibUSB> 80 06 02 03 09 04 ff 00
LibUSB< 12 03 65 00 79 00 65 00 2d 00 6f 00 6e 00 65 00 20 00
gadgetfs: 1 events received
LibUSB> 00 09 00 00 00 00 00 00
LibUSB<

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xb5ce6460 (LWP 5650)]
0xb6cead6c in HostProxy_GadgetFS::setConfig (this=0x16998, fs_cfg=0x0, hs_cfg=0x0, hs=false) at /home/debian/USBProxy/src/Plugins/Hosts/HostProxy_GadgetFS.cpp:453
453             __u8 ifc_count=fs_cfg->get_descriptor()->bNumInterfaces;
(gdb)
```

Any suggestion on how to solve this problem?

P.S.: the device is perfectly working when connected directly to the Windows PC.

tuxuser, 2016-03-26:
Problem: Same timeout problem as hllhll
OS-image: ShmooCon 2014-03
USBProxy version: 021e810 (01-03-2016)

**lsusb -v**
`Bus 002 Device 002: ID 4321:0123 *redacted*
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x4321 *redacted*
  idProduct          0x0123
  bcdDevice            1.10
  iManufacturer           1 *redacted*
  iProduct                2 *redacted*
  iSerial                 3 *redacted*
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           46
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           4
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
[... 326 lines cut ...]
Transfer error on EP83 (xfertype 2): Operation timed out
Transfer error on EP81 (xfertype 2): Operation timed out
Transfer error on EP82 (xfertype 2): Operation timed out
Transfer error on EP83 (xfertype 2): Operation timed out
Transfer error on EP81 (xfertype 2): Operation timed out
Transfer error on EP82 (xfertype 2): Operation timed out
Transfer error on EP83 (xfertype 2): Operation timed out
Transfer error on EP81 (xfertype 2): Operation timed out
Received SIGINT, stopping relaying...
Finished setup writer thread (2107) for EP00.
Finished writer thread (2110) for EP81.
Finished writer thread (2112) for EP01.
Finished writer thread (2114) for EP82.
Finished writer thread (2116) for EP83.
Finished setup reader thread (2106) for EP00.
Transfer error on EP82 (xfertype 2): Operation timed out
Finished reader thread (2113) for EP82.
Transfer error on EP83 (xfertype 2): Operation timed out
Finished reader thread (2115) for EP83.
Finished reader thread (2111) for EP01.
Transfer error on EP81 (xfertype 2): Operation timed out
Finished reader thread (2109) for EP81.
Exiting
done
`

dominicgs, 2016-03-29:
I don't know what's going on here, but it's interesting that the errors come from the IN endpoints but not the OUT EP.

lindi2, 2016-05-20:
At least

Bus 002 Device 043: ID 05ac:1402 Apple, Inc. Ethernet Adapter [A1277]

connected to Debian 8 laptop works here with git 021e810a88ea81e37c5e252f371bf2cf7173b47a. It seems to take some time to stabilize but then it seems to be fast:

```
ping 10.8.4.1
PING 10.8.4.1 (10.8.4.1) 56(84) bytes of data.
From 10.8.4.2 icmp_seq=1 Destination Host Unreachable
From 10.8.4.2 icmp_seq=2 Destination Host Unreachable
From 10.8.4.2 icmp_seq=3 Destination Host Unreachable
From 10.8.4.2 icmp_seq=4 Destination Host Unreachable
From 10.8.4.2 icmp_seq=5 Destination Host Unreachable
From 10.8.4.2 icmp_seq=6 Destination Host Unreachable
64 bytes from 10.8.4.1: icmp_seq=7 ttl=64 time=2003 ms
64 bytes from 10.8.4.1: icmp_seq=8 ttl=64 time=995 ms
64 bytes from 10.8.4.1: icmp_seq=9 ttl=64 time=0.955 ms
64 bytes from 10.8.4.1: icmp_seq=10 ttl=64 time=1.07 ms
64 bytes from 10.8.4.1: icmp_seq=11 ttl=64 time=1.46 ms
64 bytes from 10.8.4.1: icmp_seq=12 ttl=64 time=1.23 ms
64 bytes from 10.8.4.1: icmp_seq=13 ttl=64 time=1.45 ms
64 bytes from 10.8.4.1: icmp_seq=14 ttl=64 time=1.29 ms
64 bytes from 10.8.4.1: icmp_seq=15 ttl=64 time=1.29 ms
64 bytes from 10.8.4.1: icmp_seq=16 ttl=64 time=1.28 ms
64 bytes from 10.8.4.1: icmp_seq=17 ttl=64 time=1.41 ms
64 bytes from 10.8.4.1: icmp_seq=18 ttl=64 time=2.11 ms
64 bytes from 10.8.4.1: icmp_seq=19 ttl=64 time=1.32 ms
64 bytes from 10.8.4.1: icmp_seq=20 ttl=64 time=1.28 ms
64 bytes from 10.8.4.1: icmp_seq=23 ttl=64 time=1.26 ms
64 bytes from 10.8.4.1: icmp_seq=24 ttl=64 time=1.33 ms
64 bytes from 10.8.4.1: icmp_seq=25 ttl=64 time=1.40 ms
64 bytes from 10.8.4.1: icmp_seq=26 ttl=64 time=1.27 ms
64 bytes from 10.8.4.1: icmp_seq=27 ttl=64 time=1.39 ms
64 bytes from 10.8.4.1: icmp_seq=28 ttl=64 time=1.25 ms
64 bytes from 10.8.4.1: icmp_seq=29 ttl=64 time=1.19 ms
^C
--- 10.8.4.1 ping statistics ---
29 packets transmitted, 21 received, +6 errors, 27% packet loss, time 28076ms
rtt min/avg/max/mdev = 0.955/144.007/2003.141/466.421 ms, pipe 3
```

```
nc 10.8.4.1 80 | pv > /dev/null 
16.3MiB 0:00:13 [1.26MiB/s] [                                     <=>    
```

straithe, 2021-11-13:
I'm sorry there hasn't been a response to this issue in a very long time. Are any of you still experiencing this issue?

closed 2021-12-12 by straithe