usb-tools/USBProxy-legacy #50: Error mounting gadgetfs / GadgetFS not connected.

potjaune, 2015-10-01:
So I am following the steps provided here
http://gimx.fr/wiki/index.php?title=Bbb_sniffer

However whenever I run the usb-mitm command I get this error:

root@beaglebone:~# usb-mitm -l
Loading plugins from /usr/local/lib/USBProxy/
vendorId=ffffffff
productId=ffffffff
cleaning up /tmp
removing 1
Made directory /tmp/gadget-oioQcM for gadget
Error mounting gadgetfs from [/tmp/gadget-oioQcM].
Error code from mount is: [No such device]
Printing Config data
        Strings: 2
                DeviceProxy: DeviceProxy_LibUSB
                HostProxy: HostProxy_GadgetFS
        Vectors: 1
                Plugins:
                        PacketFilter_StreamLog
Pointer: 1
                PacketFilter_StreamLog::file: 0xb6d82540
Device: 12 01 10 01 00 00 00 08 6d 04 1c c3 00 64 01 02 00 01
  Manufacturer: Logitech
  Product:      USB Keyboard
        *Config(1): 09 02 3b 00 02 01 03 a0 2d
           Name: U64.00_B0001
                Interface(0):
                        *Alt(0): 09 04 00 00 01 03 01 01 02
                           Name: USB Keyboard
                                HID: 09 21 10 01 00 01 22 41 00
                                EP(81): 07 05 81 03 08 00 0a
                Interface(1):
                        *Alt(0): 09 04 01 00 01 03 00 00 02
                           Name: USB Keyboard
                                HID: 09 21 10 01 00 01 22 9f 00
                                EP(82): 07 05 82 03 04 00 ff
searching in [/tmp/gadget-oioQcM]
/tmp/gadget-oioQcM device file not found.
Error, unable to find gadget file.
Fail on open 11 Resource temporarily unavailable
GadgetFS not connected.

I redid all the instruction three times and it still gave me the same error. I am unfamiliar with how GadgetFS works unfortunately and when I try to mount by typing this line "mount -t gadgetfs gadgetfs /dev/gadget" I get his output: mount: unknown filesystem type 'gadgetfs'. 

Thanks

dominicgs, 2015-10-01:
Are you using the usbproxy image attached to the last release?

potjaune, 2015-10-01:
Are you talking about this one? 
https://github.com/dominicgs/USBProxy/releases/tag/2014-03-R1
I tried it and it gives me too many issues which I gave up on. 
This include: device not being detected via usb so I had to connect through serial console, unable to make the github project, etc..

dominicgs, 2015-10-02:
It looks like gadgetfs isn't available on your BBB, if you run `modprobe gadgetfs` does `lsmod` show it? You will also need to build the gadgetfs module with the patch provided in the doc directory.

The image already has this pre-built, which is why I recommend using it.

potjaune, 2015-10-05:
I gave the image another go, I got it to work. 
I appreciate the help, thanks!

dominicgs, 2015-10-05:
That's great, thank you for letting me know.  I'll close this issue.

closed 2015-10-05 by dominicgs

marcosox, 2016-06-07:
Sorry for re-opening this issue but I have the same exact problem.
Using the latest debian image instead (debian 8.4 from bbb site, with kernel 4.4.9-ti-r25), I can compile correctly USBProxy and (after a lot of pain regarding module_layout and MAGICVER) crosscompile the hand-patched (since file lines are no longer the same) gadgetfs.ko module (that is now located in the drivers/usb/gadget/legacy folder), but it still displays all the messages about gadgetfs that potjaune reported.
Also, I can't boot from the sd release image, so I would like to know what potjaune did in order to get it to work.
Thank you

jishnuvijay1991, 2018-05-29:
I am also facing some similar issue.
I am trying to enable usbProxy for Marvell PCB.
```
l# usb-mitm -l
Loading plugins from /usr/lib/USBProxy/
vendorId=ffffffff
productId=ffffffff
cleaning up /tmp
removing 0
Made directory /tmp/gadget-e3CXsb for gadget
Printing Config data
        Strings: 2
                DeviceProxy: DeviceProxy_LibUSB
                HostProxy: HostProxy_GadgetFS
        Vectors: 1
                Plugins:
                        PacketFilter_StreamLog
Pointer: 1
                PacketFilter_StreamLog::file: 0x7f9b5ef3a8
Device: 12 01 00 03 ef 02 01 09 10 1e 00 33 00 00 01 02 03 01
  Manufacturer: Point Grey Research
  Product:      Grasshopper3 GS3-U3-28S5C
  Serial:       01141CD2
        *Config(1): 09 02 74 00 03 01 00 80 70
                Interface(0):
                        *Alt(0): 09 04 00 00 02 ef 05 00 00
                        Other(0b): 08 0b 00 03 ef 05 00 04
                        Other(24): 14 24 01 00 00 01 00 00 00 01 00 05 06 07 08 09 0a 0b 0c 0c
                        Other(30): 06 30 00 00 00 00
                        Other(30): 06 30 00 00 00 00
                                EP(02): 07 05 02 02 00 04 00
                                EP(82): 07 05 82 02 00 04 00
                Interface(1):
                        *Alt(0): 09 04 01 00 01 ef 05 01 00
                        Other(30): 06 30 00 00 00 00
                                EP(83): 07 05 83 02 00 04 00
                Interface(2):
                        *Alt(0): 09 04 02 00 01 ef 05 02 00
                        Other(30): 06 30 0f 00 00 00
                                EP(81): 07 05 81 02 00 04 00
HS Qualifier: 0a 06 10 02 00 00 00 40 01 00
         Config(2): 09 07 12 00 01 02 00 80 fa
                Interface(0):
                        *Alt(0): 09 04 00 00 00 ff 00 00 00
searching in [/tmp/gadget-e3CXsb]
/tmp/gadget-e3CXsb device file not found.
Error, unable to find gadget file.
Fail on open 11 Resource temporarily unavailable
GadgetFS not connected.
done
```
I had loaded both musb driver and gadgetFs
```
# lsmod 
Module                  Size  Used by
musb_hdrc              73027  0
gadgetfs               18305  1

```

roshanpty, 2018-07-10:
@potjaune I know it is a little too late to ask. But how did you manage to resolve the connectivity issue of the image? I tried flashing the release image. But once I do that the BeagleBone is not reachable over SSH.

WhileE, 2018-07-30:
Console cable.