usb-tools/USBProxy-legacy #51: Is it support printers ?

taoyuan, 2015-10-05:
I have installed and tested the awesome project. It work fine with keyboard and mouse, although a little slow. 

However, it is not so lucky with thermal printers. Here is the output:

```sh
Loading plugins from /usr/local/lib/USBProxy/
vendorId=ffffffff
productId=ffffffff
cleaning up /tmp
removing 2
Made directory /tmp/gadget-NP8yI5 for gadget
Printing Config data
        Strings: 2
                DeviceProxy: DeviceProxy_LibUSB
                HostProxy: HostProxy_GadgetFS
        Vectors: 1
                Plugins:
                        PacketFilter_StreamLog
Pointer: 1
                PacketFilter_StreamLog::file: 0xb6dd69f0
Device: 12 01 00 02 00 00 00 40 83 04 20 57 00 01 01 02 03 01
  Manufacturer: Spirit
  Product:      SPRT Printer
  Serial:       11101800002
        *Config(1): 09 02 20 00 01 01 00 c0 01
                Interface(0):
                        *Alt(0): 09 04 00 00 02 07 01 02 00
                                EP(01): 07 05 01 02 40 00 00
                                EP(82): 07 05 82 02 40 00 00
searching in [/tmp/gadget-NP8yI5]
Starting setup reader thread (23002) for EP00.
Starting setup writer thread (23003) for EP00.
[80 06 02 03 09 04 02 00]
[80 06 02 03 09 04 02 00]: 1a 03
[80 06 02 03 09 04 1a 00]
[80 06 02 03 09 04 1a 00]: 1a 03 53 00 50 00 52 00 54 00 20 00 50 00 72 00 69 00 6e 00 74 00 65 00 72 00
[80 06 01 03 09 04 02 00]
[80 06 01 03 09 04 02 00]: 0e 03
[80 06 01 03 09 04 0e 00]
[80 06 01 03 09 04 0e 00]: 0e 03 53 00 70 00 69 00 72 00 69 00 74 00
[80 06 03 03 09 04 02 00]
[80 06 03 03 09 04 02 00]: 18 03
[80 06 03 03 09 04 18 00]
[80 06 03 03 09 04 18 00]: 18 03 31 00 31 00 31 00 30 00 31 00 38 00 30 00 30 00 30 00 30 00 32 00
[00 09 01 00 00 00 00 00]
Opened EP01
Opened EP82
Starting reader thread (23007) for EP82.
Starting writer thread (23008) for EP82.
Starting writer thread (23006) for EP01.
Starting reader thread (23005) for EP01.
[80 06 02 03 09 04 02 00]
[80 06 02 03 09 04 02 00]: 1a 03
[80 06 02 03 09 04 1a 00]
[80 06 02 03 09 04 1a 00]: 1a 03 53 00 50 00 52 00 54 00 20 00 50 00 72 00 69 00 6e 00 74 00 65 00 72 00
[80 06 01 03 09 04 02 00]
[80 06 01 03 09 04 02 00]: 0e 03
[80 06 01 03 09 04 0e 00]
[80 06 01 03 09 04 0e 00]: 0e 03 53 00 70 00 69 00 72 00 69 00 74 00
[80 06 03 03 09 04 02 00]
[80 06 03 03 09 04 02 00]: 18 03
[80 06 03 03 09 04 18 00]
[80 06 03 03 09 04 18 00]: 18 03 31 00 31 00 31 00 30 00 31 00 38 00 30 00 30 00 30 00 30 00 32 00
[80 06 00 03 00 00 ff 00]
[80 06 00 03 00 00 04 00]: 04 03 09 04
[80 06 02 03 09 04 ff 00]
[80 06 02 03 09 04 1a 00]: 1a 03 53 00 50 00 52 00 54 00 20 00 50 00 72 00 69 00 6e 00 74 00 65 00 72 00
[80 06 01 03 09 04 ff 00]
[80 06 01 03 09 04 0e 00]: 0e 03 53 00 70 00 69 00 72 00 69 00 74 00
[80 06 03 03 09 04 ff 00]
[80 06 03 03 09 04 18 00]: 18 03 31 00 31 00 31 00 30 00 31 00 38 00 30 00 30 00 30 00 30 00 32 00
[00 09 01 00 00 00 00 00]
// blocked here
```

And after a few seconds, it exit without any further message. Before the `usb-mitm` process exit, the host can recognize the device, but can not operate it.

Thanks for any help.

taoyuan, 2015-10-05:
`dmesg` has messages:
```
...
[11627.851557] Out of memory: Kill process 23000 (usb-mitm) score 651 or sacrifice child
[11627.862459] Killed process 23000 (usb-mitm) total-vm:402800kB, anon-rss:339576kB, file-rss:424kB
[11628.103201] gadgetfs: disconnected
```

dominicgs, 2015-10-05:
This is a known issue (it comes up about once per month).  There is a workaround included with recent git commits.  Are you running the latest code?

taoyuan, 2015-10-06:
@dominicgs  Thanks for your reply. I think I'm running the latest code. I cloned the code directly form `USBProxy` github master branch yesterday.

dominicgs, 2015-10-06:
You should be hitting this code long before you run out of memory: https://github.com/dominicgs/USBProxy/blob/45191b7269b1c9dea10aa5fa9cbdad73b34dddd2/src/lib/SafeQueue.hpp#L31

Can you run usb-mitm with the -d or -dd option?

taoyuan, 2015-10-06:
OK. Here is the full output for -dd option:

```
Version ShmooCon_2015-75-g9b1e
Running under kernel 4.1.10-bone16
CP: String DeviceProxy = DeviceProxy_LibUSB
CP: String HostProxy = HostProxy_GadgetFS
Loading plugins from /usr/local/lib/USBProxy/
vendorId=ffffffff
productId=ffffffff
cleaning up /tmp
removing 2
Made directory /tmp/gadget-8EUNgC for gadget
Printing Config data
        Strings: 2
                DeviceProxy: DeviceProxy_LibUSB
                HostProxy: HostProxy_GadgetFS
        Vectors: 1
                Plugins:
                        PacketFilter_StreamLog
Pointer: 1
                PacketFilter_StreamLog::file: 0xb6d599f0
Connected to device: 0483:5720@06 Spirit - SPRT Printer

LibUSB> 80 06 00 01 00 00 12 00
LibUSB< 12 01 00 02 00 00 00 40 83 04 20 57 00 01 01 02 03 01
LibUSB> 80 06 00 03 00 00 08 00
LibUSB< 04 03 09 04
LibUSB> 80 06 01 03 09 04 08 00
LibUSB< 0e 03 53 00 70 00 69 00
LibUSB> 80 06 01 03 09 04 0e 00
LibUSB< 0e 03 53 00 70 00 69 00 72 00 69 00 74 00
LibUSB> 80 06 02 03 09 04 08 00
LibUSB< 1a 03 53 00 50 00 52 00
LibUSB> 80 06 02 03 09 04 1a 00
LibUSB< 1a 03 53 00 50 00 52 00 54 00 20 00 50 00 72 00 69 00 6e 00 74 00 65 00 72 00
LibUSB> 80 06 03 03 09 04 08 00
LibUSB< 18 03 31 00 31 00 31 00
LibUSB> 80 06 03 03 09 04 18 00
LibUSB< 18 03 31 00 31 00 31 00 30 00 31 00 38 00 30 00 30 00 30 00 30 00 32 00
LibUSB> 80 06 00 02 00 00 09 00
LibUSB< 09 02 20 00 01 01 00 c0 01
LibUSB> 80 06 00 02 00 00 20 00
LibUSB< 09 02 20 00 01 01 00 c0 01 09 04 00 00 02 07 01 02 00 07 05 01 02 40 00 00 07 05 82 02 40 00 00
LibUSB> 80 06 00 06 00 00 0a 00
Error sending setup packet: Pipe error
LibUSB> 80 08 00 00 00 00 01 00
LibUSB< 01
Device: 12 01 00 02 00 00 00 40 83 04 20 57 00 01 01 02 03 01
  Manufacturer: Spirit
  Product:      SPRT Printer
  Serial:       11101800002
        *Config(1): 09 02 20 00 01 01 00 c0 01
                Interface(0):
                        *Alt(0): 09 04 00 00 02 07 01 02 00
                                EP(01): 07 05 01 02 40 00 00
                                EP(82): 07 05 82 02 40 00 00

        00 00 00 00 09 02 20 00 01 01 00 c0 01 09 04 00 00 02 07 01 02 00 07 05 01 02 40 00 00 07 05 82
        02 40 00 00 09 02 20 00 01 01 00 c0 01 09 04 00 00 02 07 01 02 00 07 05 01 02 40 00 00 07 05 82
        02 40 00 00 12 01 00 02 00 00 00 40 83 04 20 57 00 01 01 02 03 01
searching in [/tmp/gadget-8EUNgC]
Starting setup reader thread (2871) for EP00.
Starting setup writer thread (2872) for EP00.
gadgetfs: 1 events received
gadgetfs: event 1
gadgetfs: 1 events received
[80 06 02 03 09 04 02 00]
LibUSB> 80 06 02 03 09 04 02 00
LibUSB< 1a 03
[... 522 lines cut ...]
[24347.785109] [  348]     0   348     7658      197       9       0        0             0 rsyslogd
[24347.799061] [  398]     0   398      887       30       5       0        0             0 agetty
[24347.812721] [  401]     0   401      842       30       6       0        0             0 agetty
[24347.826349] [  405]     0   405     1579      318       8       0        0             0 apache2
[24347.839984] [  513]     0   513     2516      163       8       0        0             0 sshd
[24347.853266] [  515]  1000   515     2516      168       8       0        0             0 sshd
[24347.866451] [  516]  1000   516     1127       91       6       0        0             0 bash
[24347.879566] [  521]     0   521     2516      163       9       0        0             0 sshd
[24347.892595] [  523]  1000   523     2516      169       7       0        0             0 sshd
[24347.905530] [  524]  1000   524     1130       94       7       0        0             0 bash
[24347.918389] [  551]  1000   551     1942      822       6       0        0             0 tmux
[24347.931174] [  552]  1000   552      989       44       5       0        0             0 bash
[24347.943872] [  553]  1000   553     1130       93       6       0        0             0 bash
[24347.956523] [  564]  1000   564      989       44       5       0        0             0 bash
[24347.969143] [  566]  1000   566     1130       94       6       0        0             0 bash
[24347.981697] [  572]  1000   572      989       44       6       0        0             0 bash
[24347.994189] [  574]  1000   574     1130       94       5       0        0             0 bash
[24348.006648] [ 2214]    33  2214    57179      383      35       0        0             0 apache2
[24348.019341] [ 2215]    33  2215    57179      383      35       0        0             0 apache2
[24348.031995] [ 2867]     0  2867     1356       83       7       0        0             0 sudo
[24348.044272] [ 2868]     0  2868   130268   114555     239       0        0             0 usb-mitm
[24348.056757] Out of memory: Kill process 2868 (usb-mitm) score 877 or sacrifice child
[24348.068012] Killed process 2868 (usb-mitm) total-vm:521072kB, anon-rss:457944kB, file-rss:276kB
[24348.305735] gadgetfs: disconnected
```

dominicgs, 2015-10-06:
This is strange.  Are you running on a BeagleBone Black?  Are you using the OS image that I published on the releases page?

taoyuan, 2015-10-06:
Yes. I'm running on a BeagleBone Black.  I have tested on the OS image that published on the releases page before, and encountered the problem. So I compiled a new OS version based on the 4.1 kernel according to the [related documents](https://eewiki.net/display/linuxonarm/BeagleBone+Black), of cause opened GadgetFS function.

The outputs above is running on the 4.1 kernel OS image.

taoyuan, 2015-10-06:
I ran once again on your OS image with the latest version. And the outputs is:

```
debian@arm:~/USBProxy/src/build$ sudo usb-mitm -l -d
Version ShmooCon_2015-76-g562f
Running under kernel 3.12.0-bone8
CP: String DeviceProxy = DeviceProxy_LibUSB
CP: String HostProxy = HostProxy_GadgetFS
Loading plugins from /usr/local/lib/USBProxy/
vendorId=ffffffff
productId=ffffffff
cleaning up /tmp
removing 0
Made directory /tmp/gadget-zL2KgX for gadget
Printing Config data
	Strings: 2
		DeviceProxy: DeviceProxy_LibUSB
		HostProxy: HostProxy_GadgetFS
	Vectors: 1
		Plugins:
			PacketFilter_StreamLog
Pointer: 1
		PacketFilter_StreamLog::file: 0xb6d0c9f0
Connected to device: 0483:5720@02 Spirit - SPRT Printer

Error sending setup packet: Pipe error
Device: 12 01 00 02 00 00 00 40 83 04 20 57 00 01 01 02 03 01
  Manufacturer: Spirit
  Product:      SPRT Printer
  Serial:       11101800002
	*Config(1): 09 02 20 00 01 01 00 c0 01
		Interface(0):
			*Alt(0): 09 04 00 00 02 07 01 02 00
				EP(01): 07 05 01 02 40 00 00
				EP(82): 07 05 82 02 40 00 00

	00 00 00 00 09 02 20 00 01 01 00 c0 01 09 04 00 00 02 07 01 02 00 07 05 01 02 40 00 00 07 05 82
	02 40 00 00 09 02 20 00 01 01 00 c0 01 09 04 00 00 02 07 01 02 00 07 05 01 02 40 00 00 07 05 82
	02 40 00 00 12 01 00 02 00 00 00 40 83 04 20 57 00 01 01 02 03 01
searching in [/tmp/gadget-zL2KgX]
Starting setup writer thread (2470) for EP00.
Starting setup reader thread (2469) for EP00.
[control] 03 00 00 00 00 00 00 00 01 00 00 00
gadgetfs: event 1
[control] 80 06 02 03 09 04 02 00 03 00 00 00
[80 06 02 03 09 04 02 00]
[80 06 02 03 09 04 02 00]: 1a 03
[control] 80 06 02 03 09 04 1a 00 03 00 00 00
[80 06 02 03 09 04 1a 00]
[80 06 02 03 09 04 1a 00]: 1a 03 53 00 50 00 52 00 54 00 20 00 50 00 72 00 69 00 6e 00 74 00 65 00 72 00
[control] 80 06 01 03 09 04 02 00 03 00 00 00
[80 06 01 03 09 04 02 00]
[80 06 01 03 09 04 02 00]: 0e 03
[control] 80 06 01 03 09 04 0e 00 03 00 00 00
[80 06 01 03 09 04 0e 00]
[80 06 01 03 09 04 0e 00]: 0e 03 53 00 70 00 69 00 72 00 69 00 74 00
[control] 80 06 03 03 09 04 02 00 03 00 00 00
[80 06 03 03 09 04 02 00]
[80 06 03 03 09 04 02 00]: 18 03
[control] 80 06 03 03 09 04 18 00 03 00 00 00
[80 06 03 03 09 04 18 00]
[80 06 03 03 09 04 18 00]: 18 03 31 00 31 00 31 00 30 00 31 00 38 00 30 00 30 00 30 00 30 00 32 00
[control] 80 06 03 03 09 04 ff 00 03 00 00 00
[80 06 03 03 09 04 ff 00]
[80 06 03 03 09 04 18 00]: 18 03 31 00 31 00 31 00 30 00 31 00 38 00 30 00 30 00 30 00 30 00 32 00
[control] 80 06 00 03 00 00 ff 00 03 00 00 00
[80 06 00 03 00 00 ff 00]
[80 06 00 03 00 00 04 00]: 04 03 09 04
[control] 80 06 02 03 09 04 ff 00 03 00 00 00
[80 06 02 03 09 04 ff 00]
[... 77 lines cut ...]
[  594.318044] [ 1887]     0  1887      907       40       5        0             0 cron
[  594.326291] [ 1917]   101  1917      718       56       4        0             0 dbus-daemon
[  594.335186] [ 1962]   102  1962      762       68       4        0             0 avahi-daemon
[  594.344163] [ 1965]   102  1965      762       55       4        0             0 avahi-daemon
[  594.353140] [ 2089]     0  2089     1379      107       4        0         -1000 sshd
[  594.361387] [ 2114]   104  2114     1270      110       5        0             0 ntpd
[  594.369634] [ 2139]     0  2139      904       32       5        0             0 getty
[  594.377973] [ 2140]     0  2140      904       32       4        0             0 getty
[  594.386310] [ 2141]     0  2141      904       32       5        0             0 getty
[  594.394649] [ 2142]     0  2142      904       32       4        0             0 getty
[  594.402986] [ 2143]     0  2143      904       32       4        0             0 getty
[  594.411323] [ 2144]     0  2144      904       32       4        0             0 getty
[  594.419666] [ 2145]     0  2145      466       31       4        0             0 getty
[  594.428005] [ 2146]     0  2146     2163      156       6        0             0 sshd
[  594.436251] [ 2148]  1000  2148     2163      155       5        0             0 sshd
[  594.444497] [ 2149]  1000  2149     1109       92       5        0             0 bash
[  594.452744] [ 2450]     0  2450     2163      156       6        0             0 sshd
[  594.460990] [ 2452]  1000  2452     2163      155       5        0             0 sshd
[  594.469234] [ 2453]  1000  2453     1109       92       4        0             0 bash
[  594.477480] [ 2464]     0  2464     1104       66       5        0             0 sudo
[  594.485726] [ 2465]     0  2465   136681   120962     250        0             0 usb-mitm
[  594.494333] Out of memory: Kill process 2465 (usb-mitm) score 924 or sacrifice child
[  594.502491] Killed process 2465 (usb-mitm) total-vm:546724kB, anon-rss:483692kB, file-rss:156kB
[  594.713704] gadgetfs: disconnected
```

dominicgs, 2015-10-06:
This is very strange.  We've already included code to work around this problem.

Could you list the version of everything?  libUSB, the git revision of the code?  Did you install the code or are you running it from the build directory?

taoyuan, 2015-10-06:
According to the readme, I compiled `USBProxy`:

```sh
mkdir src/build
cd src/build
cmake ..
make
sudo make install
sudo ldconfig
```

And run with `sudo usb-mitm -l -d`

Here is the code version info:

```
debian@arm:~/USBProxy$ git remote show origin
* remote origin
  Fetch URL: https://github.com/dominicgs/USBProxy.git
  Push  URL: https://github.com/dominicgs/USBProxy.git
  HEAD branch: master
  Remote branch:
    master tracked
  Local branch configured for 'git pull':
    master merges with remote master
  Local ref configured for 'git push':
    master pushes to master (fast-forwardable)
```
```
debian@arm:~/USBProxy$ git log -n 1
commit 562f1691abf2fddc9aada5caf18e5f9baf2c3491
Author: Tao Yuan <towyuan@outlook.com>
Date:   Tue Oct 6 11:45:11 2015 +0000

    Revert "Update compiler for travis build (still won't build though)"

    This reverts commit 9b1e83c59fa646baf0ba2d9c99db06873a493cb4.
```

This is `cmake` outputs includes some versions info:

```
debian@arm:~/USBProxy/src/build$ cmake ..
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26")
-- checking for module 'libusb-1.0'
--   found libusb-1.0, version 1.0.17
-- Found LIBUSB: /usr/local/lib/libusb-1.0.so
-- Could NOT find LORCON (missing:  LORCON_LIBRARIES LORCON_INCLUDE_DIR)
-- Pcap include dirs set to /usr/include
-- Pcap library set to  /usr/lib/arm-linux-gnueabihf/libpcap.so
-- Looking for pcap_version
-- Looking for pcap_version - found
-- Looking for pcap_open_dead
-- Looking for pcap_open_dead - found
-- Looking for pcap_freecode
-- Looking for pcap_freecode - found
-- Looking for pcap_breakloop
-- Looking for pcap_breakloop - found
-- Looking for pcap_create
-- Looking for pcap_create - found
-- Looking for pcap_datalink_name_to_val
-- Looking for pcap_datalink_name_to_val - found
-- Looking for pcap_datalink_val_to_description
-- Looking for pcap_datalink_val_to_description - found
-- Looking for pcap_datalink_val_to_name
-- Looking for pcap_datalink_val_to_name - found
-- Looking for pcap_findalldevs
-- Looking for pcap_findalldevs - found
-- Looking for pcap_free_datalinks
-- Looking for pcap_free_datalinks - found
-- Looking for pcap_get_selectable_fd
-- Looking for pcap_get_selectable_fd - found
-- Looking for pcap_lib_version
-- Looking for pcap_lib_version - found
-- Looking for pcap_list_datalinks
-- Looking for pcap_list_datalinks - found
-- Looking for pcap_set_datalink
-- Looking for pcap_set_datalink - found
-- Looking for pcap_open
-- Looking for pcap_open - not found
-- Looking for pcap_findalldevs_ex
-- Looking for pcap_findalldevs_ex - not found
-- Looking for pcap_createsrcstr
-- Looking for pcap_createsrcstr - not found
-- Found PCAP: /usr/include
-- Configuring done
-- Generating done
-- Build files have been written to: /home/debian/USBProxy/src/build
```

devilicecream, 2015-10-23:
Up for this issue. I'm also testing the project with a thermal printer and I'm getting a similar issue.
Followed the instructions here: http://gimx.fr/wiki/index.php?title=Bbb_sniffer (not including the system, which I downloaded from the Releases section on here).
libUSB version: 1.0.19.
dmesg:
```
[ 1349.020545] usb 2-1: new full-speed USB device number 5 using musb-hdrc
[ 1349.157656] usb 2-1: device v0519 p0003 is not supported
[ 1349.163293] usb 2-1: New USB device found, idVendor=0519, idProduct=0003
[ 1349.170352] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1349.177863] usb 2-1: Product: Star TSP143ECO
[ 1349.182361] usb 2-1: Manufacturer: STAR
[ 1349.209558] usblp 2-1:1.0: usblp0: USB Bidirectional printer dev 5 if 0 alt 0 proto 2 vid 0x0519 pid 0x0003
[ 1349.222020] usbcore: registered new interface driver usblp
[ 1367.562962] nop musb-hdrc.0.auto: failed to start (null): -120
[ 1367.601592] usblp0: removed
[ 1367.610329] gadgetfs: bound to musb-hdrc driver
[ 1367.912740] gadgetfs: connected
[ 1367.920955] gadgetfs: disconnected
[ 1367.998701] gadgetfs: connected
[ 1368.299982] gadgetfs: suspended from state 3
[ 1368.440361] gadgetfs: disconnected
[ 1368.479796] gadgetfs: connected
[ 1368.487973] gadgetfs: disconnected
[ 1368.567763] gadgetfs: connected
[ 1368.601467] gadgetfs: configuration #1
[ 1394.119265] usblp 2-1:1.0: usblp0: USB Bidirectional printer dev 5 if 0 alt 0 proto 2 vid 0x0519 pid 0x0003
[ 1394.132400] gadgetfs: disconnected
```
I see some weird stuff at 1367.562962 here.
sudo usb-mitm -ld

```
Version ShmooCon_2015-79-g6cc9
Running under kernel 3.12.0-bone8
CP: String DeviceProxy = DeviceProxy_LibUSB
CP: String HostProxy = HostProxy_GadgetFS
Loading plugins from /usr/local/lib/USBProxy/
vendorId=ffffffff
productId=ffffffff
cleaning up /tmp
removing 0
Made directory /tmp/gadget-T08SYT for gadget
Printing Config data
	Strings: 2
		DeviceProxy: DeviceProxy_LibUSB
		HostProxy: HostProxy_GadgetFS
	Vectors: 1
		Plugins:
			PacketFilter_StreamLog
Pointer: 1
		PacketFilter_StreamLog::file: 0xb6d50540
Connected to device: 0519:0003@05 STAR - Star TSP143ECO

Error sending setup packet: Pipe error
Device: 12 01 00 02 00 00 00 40 19 05 03 00 00 01 01 02 00 01
  Manufacturer: STAR
  Product:      Star TSP143ECO
	*Config(1): 09 02 20 00 01 01 00 c0 32
		Interface(0):
			*Alt(0): 09 04 00 00 02 07 01 02 00
				EP(81): 07 05 81 02 40 00 00
				EP(02): 07 05 02 02 40 00 00

	00 00 00 00 09 02 20 00 01 01 00 c0 32 09 04 00 00 02 07 01 02 00 07 05 81 02 40 00 00 07 05 02
	02 40 00 00 09 02 20 00 01 01 00 c0 32 09 04 00 00 02 07 01 02 00 07 05 81 02 40 00 00 07 05 02
	02 40 00 00 12 01 00 02 00 00 00 40 19 05 03 00 00 01 01 02 00 01
searching in [/tmp/gadget-T08SYT]
Starting setup writer thread (7834) for EP00.
Starting setup reader thread (7833) for EP00.
gadgetfs: event 1
gadgetfs: event 2
gadgetfs: event 1
[80 06 00 03 00 00 ff 00]
[80 06 00 03 00 00 04 00]: 04 03 09 04
[80 06 02 03 09 04 ff 00]
[80 06 02 03 09 04 1e 00]: 1e 03 53 00 74 00 61 00 72 00 20 00 54 00 53 00 50 00 31 00 34 00 33 00 45 00 43 00 4f 00
gadgetfs: event 4
gadgetfs: event 2
gadgetfs: event 1
gadgetfs: event 2
gadgetfs: event 1
[80 06 00 03 00 00 ff 00]
[80 06 00 03 00 00 04 00]: 04 03 09 04
[80 06 02 03 09 04 ff 00]
[80 06 02 03 09 04 1e 00]: 1e 03 53 00 74 00 61 00 72 00 20 00 54 00 53 00 50 00 31 00 34 00 33 00 45 00 43 00 4f 00
[00 09 01 00 00 00 00 00]
Opened EP81
Opened EP02
Sending ACK
Starting writer thread (7839) for EP02.
Starting reader thread (7838) for EP02.
Starting writer thread (7837) for EP81.
Starting reader thread (7836) for EP81.
81[9]: 23 06 00 00 00 00 00 00 00
81[9]: 23 06 00 00 00 00 00 00 00
81[9]: 23 06 00 00 00 00 00 00 00
Warning: queue fills up! Feel free to search the bug in either the driver or usbproxy.
81[9]: 23 06 00 00 00 00 00 00 00
81[9]: 23 06 00 00 00 00 00 00 00
81[9]: 23 06 00 00 00 00 00 00 00
81[9]: 23 06 00 00 00 00 00 00 00
[ This continues forever ....]
^CReceived SIGINT, stopping relaying...
Finished setup writer thread (7834) for EP00.
Finished writer thread (7839) for EP02.
Finished reader thread (7836) for EP81.
Finished setup reader thread (7833) for EP00.
Finished writer thread (7837) for EP81.
Finished reader thread (7838) for EP02.
Exiting
```

devilicecream, 2015-10-23:
Also, running ```lsmod``` gives this output
```
Module                  Size  Used by
gadgetfs               15332  2147483647 
snd_soc_omap            2834  0 
snd_pcm_dmaengine       6073  1 snd_soc_omap
snd_soc_core          119143  1 snd_soc_omap
snd_compress            7406  1 snd_soc_core
regmap_spi              1901  1 snd_soc_core
snd_pcm                76780  3 snd_soc_core,snd_soc_omap,snd_pcm_dmaengine
snd_page_alloc          5132  1 snd_pcm
snd_timer              18670  1 snd_pcm
snd                    59355  4 snd_soc_core,snd_timer,snd_pcm,snd_compress
soundcore               6798  1 snd
usblp                  10407  0 
smsc                    2469  0 
```
Which I found very strange O.O

devilicecream, 2015-10-31:
@dominicgs any thoughts on this one?

dominicgs, 2015-10-31:
I'm not sure I see the problem in the output from usb-mitm.  Is it not passing through traffic to the printer?

You may want to blacklist the usblp module so that the host doesn't try to claim it, but that doesn't appear to be causing a problem.

devilicecream, 2015-11-02:
The traffic is indeed not being relayed, the problem I see in the log is this line:
```
Warning: queue fills up! Feel free to search the bug in either the driver or usbproxy.
```
Looks like the queue doesn't get freed because it can't write on the output port. Any idea of why this happens?

dominicgs, 2015-11-02:
USBProxy appears to be polling one of the EPs on the printer, but the host isn't polling USBProxy at the same rate.  Dropping these packets when the queue is full stops us from running out of memory and appeared to work for other users who have tried this with printers.

Maybe we need to use a smarter approach, we could stop polling the EP once there are X packets in the queue.  However, I'm not sure that is going to solve your problem because you're not getting any traffic passed on the other EPs.

taoyuan, 2015-11-06:
It sounds like reasonable about the memory issue. 

But the main problem is `block`. I tried to send a print instruction before queue is full and exiting, usb-mitman has no response. It seems like that gadgetfs has not transfer all requests from host to device including data of endpoints.

dominicgs, 2015-11-06:
Yes, that's strange.  I would expect data going to the printer to be sent without problem.  Have you tried using `usbmon` on the BBB to see what is being sent to the printer?

taoyuan, 2015-11-06:
I haven't tried using `usbmon`. 

But I tried using `node.js` to implement similar functionality, and found that the printer can not process control request `[00 09 01 00 00 00 00 00]`, it throws some `transfer error`.

devilicecream, 2015-11-12:
Another two cents on this one. I'm trying to figure out what throws the ```Error sending setup packet: Pipe error``` at the start, and I found this code in ```DeviceQualifier.cpp```
```
DeviceQualifier::DeviceQualifier(Device* _device,DeviceProxy* proxy) {
        device=_device;
        usb_ctrlrequest setup_packet;
        setup_packet.bRequestType=USB_DIR_IN | USB_TYPE_STANDARD | USB_RECIP_DEVICE;
        setup_packet.bRequest=USB_REQ_GET_DESCRIPTOR;
        setup_packet.wValue=USB_DT_DEVICE_QUALIFIER<<8;
        setup_packet.wIndex=0;
        setup_packet.wLength=10;
        int len=0;
        if (proxy->control_request(&setup_packet,&len,(__u8 *)&descriptor)) {
               descriptor.bLength=0;
               configurations=NULL;
               return;
        }
```
The value (decimal) sent to the device are:
```
bRequestType: 128 bRequest: 6 wValue: 1536 wIndex: 0 wLength: 10 timeout: 500
````
And I found this  http://www.keil.com/support/man/docs/rlarm/rlarm_usb_descript_dev_qualifier.htm, saying that if a full-speed only device receives a GetDescriptor() request for a device_qualifier, it must respond with a request error. Then, the host must not make a request for an other_speed_configuration descriptor.
Therefore, is it the Pipe error ok in the case of a full-speed only device? Should it be handled better?

dominicgs, 2015-11-12:
That seems like a reasonable suggestion, I think the error should definitely be handled cleanly.

I can't exactly remember the situation, but I seem to remember that there is something strange about the way that gadgetfs or musb-hdrc handles full/high speed devices.  I believe it always requires a second set of descriptors to present itself as a high speed device, but I can't remember why.

palesius, 2015-11-12:
This is equally vague. But I think gadgetfs doesn't let us directly respond
to the initial descriptor request. We need to provide both and it handles
it all behind the scenes, depending on what is requested.

straithe, 2021-11-13:
Sorry there hasn't been a response to this issue in a while. @taoyuan do you still need assistance?

closed 2021-12-12 by straithe