usb-tools/USBProxy-legacy #58: usbproxy-fd-umass.py - Heap Corruption

bub0, 2016-02-22:
When I attempt to use this script (with multiple FS images I've created) I always get varying memory corruption errors. This is using the image provided on the release page.

I have tried with the following FS images:

50M ext3
50M ext4
50M NTFS

The errors seem to occur directly after the mass storage transfer takes place.

Are there any FS images that have been used with this script that I can test with to get a baseline?

lindi2, 2016-05-20:
I can reproduce this with git 021e810a88ea81e37c5e252f371bf2cf7173b47a and it occurs without any file system at all:

```
dd if=/dev/zero of=hdd.img bs=1M count=50
./bindings/python/usbproxy-fd-umass.py
...
USB mass storage interface handling 31 bytes of SCSI data
USB mass storage interface got SCSI Read (10), lba 0 + 8 block(s)
9
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
8
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
*** glibc detected *** python3: double free or corruption (!prev): 0x0026c330 ***
```
*** glibc detected *** python3: double free or corruption (!prev): 0x0038b2e8 ***

lindi2, 2016-05-20:
Backtrace:
```
(gdb) thread apply all bt

Thread 9 (Thread 0xb0720470 (LWP 10977)):
#0  __libc_do_syscall () at ../ports/sysdeps/unix/sysv/linux/arm/eabi/libc-do-syscall.S:44
#1  0xb6fcd0bc in __libc_pread (fd=<optimized out>, buf=<optimized out>, count=<optimized out>, offset=0) at ../ports/sysdeps/unix/sysv/linux/arm/eabi/pread.c:53
#2  0xb6a56b64 in handle_fildes_io (arg=<optimized out>) at ../sysdeps/pthread/aio_misc.c:536
#3  0xb6fc6ebc in start_thread (arg=0xb0720470) at pthread_create.c:306
#4  0xb6eb3328 in ?? () at ../ports/sysdeps/unix/sysv/linux/arm/nptl/../clone.S:116 from /lib/arm-linux-gnueabihf/libc.so.6
#5  0xb6eb3328 in ?? () at ../ports/sysdeps/unix/sysv/linux/arm/nptl/../clone.S:116 from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 8 (Thread 0xb0f20470 (LWP 10976)):
#0  __libc_do_syscall () at ../ports/sysdeps/unix/sysv/linux/arm/eabi/libc-do-syscall.S:44
#1  0xb6e42f6a in __GI_raise (sig=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:67
#2  0xb6e45408 in __GI_abort () at abort.c:92
#3  0xb6e6a2d6 in __libc_message (do_abort=2, fmt=0xb6eecc4c "*** glibc detected *** %s: %s: 0x%s ***\n") at ../sysdeps/unix/sysv/linux/libc_fatal.c:189
#4  0xb6e70faa in malloc_printerr (action=<optimized out>, str=0xb6eece6c "double free or corruption (!prev)", ptr=<optimized out>) at malloc.c:6312
#5  0xb6e741ca in __GI___libc_free (mem=<optimized out>) at malloc.c:3738
#6  0xb6aab412 in Packet::~Packet (this=0x313a70, __in_chrg=<optimized out>) at /home/debian/USBProxy/src/lib/Packet.h:24
#7  0xb6aaccb0 in __gnu_cxx::new_allocator<Packet>::destroy<Packet> (this=0x313a6c, __p=0x313a70) at /usr/include/c++/4.7/ext/new_allocator.h:114
#8  0xb6aacc3e in std::allocator_traits<std::allocator<Packet> >::_S_destroy<Packet> (__a=..., __p=0x313a70) at /usr/include/c++/4.7/bits/alloc_traits.h:279
#9  0xb6aacbce in std::allocator_traits<std::allocator<Packet> >::destroy<Packet> (__a=..., __p=0x313a70) at /usr/include/c++/4.7/bits/alloc_traits.h:402
#10 0xb6aaca0a in std::_Sp_counted_ptr_inplace<Packet, std::allocator<Packet>, (__gnu_cxx::_Lock_policy)2>::_M_dispose (this=0x313a60)
    at /usr/include/c++/4.7/bits/shared_ptr_base.h:410
#11 0xb6aa6366 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x313a60) at /usr/include/c++/4.7/bits/shared_ptr_base.h:147
#12 0xb6aa59ea in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0xb0f1fd24, __in_chrg=<optimized out>)
    at /usr/include/c++/4.7/bits/shared_ptr_base.h:558
#13 0xb6aab544 in std::__shared_ptr<Packet, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0xb0f1fd20, __in_chrg=<optimized out>)
    at /usr/include/c++/4.7/bits/shared_ptr_base.h:813
#14 0xb6aababc in std::__shared_ptr<Packet, (__gnu_cxx::_Lock_policy)2>::operator=(std::__shared_ptr<Packet, (__gnu_cxx::_Lock_policy)2>&&) (this=0xb0f1fd58, __r=...)
    at /usr/include/c++/4.7/bits/shared_ptr_base.h:900
#15 0xb6aab7d4 in std::shared_ptr<Packet>::operator=(std::shared_ptr<Packet>&&) (this=0xb0f1fd58, __r=...) at /usr/include/c++/4.7/bits/shared_ptr.h:292
#16 0xb6ab3f16 in RelayWriter::relay_write (this=0x2a2578) at /home/debian/USBProxy/src/lib/RelayWriter.cpp:134
#17 0xb6aaab02 in std::_Mem_fn<void (RelayWriter::*)()>::operator() (this=0x26c340, __object=0x2a2578) at /usr/include/c++/4.7/functional:554
#18 0xb6aaa8bc in std::_Bind_simple<std::_Mem_fn<void (RelayWriter::*)()> (RelayWriter*)>::_M_invoke<0u>(std::_Index_tuple<0u>) (this=0x26c33c)
    at /usr/include/c++/4.7/functional:1598
#19 0xb6aaa734 in std::_Bind_simple<std::_Mem_fn<void (RelayWriter::*)()> (RelayWriter*)>::operator()() (this=0x26c33c) at /usr/include/c++/4.7/functional:1586
#20 0xb6aaa63c in std::thread::_Impl<std::_Bind_simple<std::_Mem_fn<void (RelayWriter::*)()> (RelayWriter*)> >::_M_run() (this=0x26c330) at /usr/include/c++/4.7/thread:115
#21 0xb6a2c6d0 in ?? () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#22 0xb6a2c6d0 in ?? () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

---Type <return> to continue, or q <return> to quit---
Thread 7 (Thread 0xb1720470 (LWP 10975)):
#0  __libc_do_syscall () at ../ports/sysdeps/unix/sysv/linux/arm/eabi/libc-do-syscall.S:43
#1  0xb6ebc5dc in __lll_lock_wait_private (futex=0xb6f03250) at ../ports/sysdeps/unix/sysv/linux/arm/nptl/lowlevellock.c:32
#2  0xb6e74248 in __GI___libc_free (mem=<optimized out>) at malloc.c:3736
#3  0x00109616 in ?? ()
#4  0x00109616 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 6 (Thread 0xb1f20470 (LWP 10974)):
#0  __libc_do_syscall () at ../ports/sysdeps/unix/sysv/linux/arm/eabi/libc-do-syscall.S:43
#1  0xb6fca02e in __pthread_cond_wait (cond=0x274ea8, mutex=0x274e90) at pthread_cond_wait.c:153
#2  0xb6a2a88c in std::condition_variable::wait(std::unique_lock<std::mutex>&) () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#3  0xb6aab76c in SafeQueue<std::shared_ptr<Packet> >::dequeue (this=0x274e68) at /home/debian/USBProxy/src/lib/SafeQueue.hpp:43
#4  0xb6ab3f06 in RelayWriter::relay_write (this=0x31a530) at /home/debian/USBProxy/src/lib/RelayWriter.cpp:134
#5  0xb6aaab02 in std::_Mem_fn<void (RelayWriter::*)()>::operator() (this=0x36de18, __object=0x31a530) at /usr/include/c++/4.7/functional:554
[... cut ...]
#7  0xb6aaa74c in std::_Bind_simple<std::_Mem_fn<void (RelayReader::*)()> (RelayReader*)>::operator()() (this=0x3937ac) at /usr/include/c++/4.7/functional:1586
#8  0xb6aaa658 in std::thread::_Impl<std::_Bind_simple<std::_Mem_fn<void (RelayReader::*)()> (RelayReader*)> >::_M_run() (this=0x3937a0) at /usr/include/c++/4.7/thread:115
#9  0xb6a2c6d0 in ?? () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#10 0xb6a2c6d0 in ?? () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

```

straithe, 2021-11-13:
I am closing this issue as the initial poster is no longer on GitHub.

closed 2021-11-13 by straithe