usb-tools/USBProxy-legacy #59: Lagg on OS X or Linux

N1KLoT, 2016-03-13:
Im running the latest release img on a beaglebone black rev c and it works well when connected to a Windows machine, but the mouse is almost unusable on Linux or OS X machines. I get a lot of "Transfer error on EP81 (xfertype 3): Operation timed out", any idea why this could be?

dominicgs, 2016-03-14:
The release image is very old, I recommend running `git pull` from the USBProxy directory and then rebuilding the project.

N1KLoT, 2016-03-15:
That was the first thing I tried, had to update gcc and g++ and then I got it working but the lagg is actually worse than before.

JesusFreke, 2016-04-21:
I have a similar problem when trying to proxy a joystick that continuously sends events at basically the maximum rate. It will receive something like 4-5 events from the joystick in the time it takes to send 1 packet to the endpoint via gadgetfs.

As best as I can tell, the problem is that in relay_write() in RelayWriter.cpp, it waits until the previous write has completed before starting the next write, which seems like a complicated way to do synchronous IO using the asynchronous functions.

If I modify it so that it doesn't wait for the previous write to complete, it is able to keep up with the stream of events coming from the joystick.

@dominicgs Is this something you would consider accepting a patch for? I'm not sure if there might be unintended side effects of doing the IO like this..

JesusFreke, 2016-04-21:
Actually, I see that the GadgetFS host proxy would probably need to be rewritten so that it properly supports having multiple async writes out at the same time.

dominicgs, 2016-04-21:
I would accept that patch.  It's worth submitting anyway as I plan to switch out GadgetFS for ConfigFS as soon as I get time to work on it.

lindi2, 2016-05-20:
With git 021e810a88ea81e37c5e252f371bf2cf7173b47a handling traffic between a logitech mouse and Debian 8 host I get 15 updates per second.

lindi2, 2016-05-20:
With

``diff --git a/src/lib/RelayWriter.cpp b/src/lib/RelayWriter.cpp
index 532f779..8bf1570 100644
--- a/src/lib/RelayWriter.cpp
+++ b/src/lib/RelayWriter.cpp
@@ -141,7 +141,7 @@ void RelayWriter::relay_write() {
                        }
                        if (p->transmit) {
                                proxy->send_data(endpoint,attributes,maxPacketSize,p->data,p->wLength);
-                               writing=true;
+                               //writing=true;
                        }
                } else {
                        writing=!(proxy->send_wait_complete(endpoint, READ_TIMEOUT_MS));
``

I get 84 updates per second.

straithe, 2021-11-13:
I'm sorry there has been no response to this in a very long time. @N1KLoT are you still experiencing this issue?

closed 2021-12-12 by straithe