usb-tools/USBProxy-legacy #62: gadgetfs does not mount

Ramanathan-R, 2016-12-03:
1) Upgraded Debian wheezy to jessie on a BBB.
2) Built USBProxy from source following instructions. Only "liborcon" was not installed. But "usb-mitm" was still built.
3) Ran "usb-mitm" which created a temporary directory as "/tmp/gadget-xxx". But mounting this as gadgetfs fails.
4) "lsmod" shows "gadgetfs".
5) Running "mount -t gadgetfs none /tmp/gadget-VKuV7c" gives output "mount: unknown filesystem type 'gadgetfs'"
6) Scratch head, do some Googling and still no solution :-)

Any ideas/suggestions/pointers/links etc. ?

marcosox, 2016-12-03:
What error message does usb-mitm return on step 3?
Also, you must run usb-mitm as root / with sudo

Ramanathan-R, 2016-12-03:
root@beaglebone:~# lsmod
Module                  Size  Used by
g_multi                50407  0
libcomposite           15028  1 g_multi
omap_rng                4062  0
nfsd                  187513  2
mt7601Usta            639170  0

root@beaglebone:~# usb-mitm -l
Loading plugins from /usr/local/lib/USBProxy/
vendorId=ffffffff
productId=ffffffff
cleaning up /tmp
removing 0
Made directory /tmp/gadget-iNvIJk for gadget
Error mounting gadgetfs from [/tmp/gadget-iNvIJk].
Error code from mount is: [No such device]
Printing Config data
        Strings: 2
                DeviceProxy: DeviceProxy_LibUSB
                HostProxy: HostProxy_GadgetFS
        Vectors: 1
                Plugins:
                        PacketFilter_StreamLog
Pointer: 1
                PacketFilter_StreamLog::file: 0xb6d239f0
Unable to connect to device proxy.
done

root@beaglebone:~# lsmod
Module                  Size  Used by
gadgetfs               12527  0
g_multi                50407  0
libcomposite           15028  1 g_multi
omap_rng                4062  0
nfsd                  187513  2
mt7601Usta            639170  0

root@beaglebone:~# ls -al /tmp
total 44
drwxrwxrwt 10 root root 4096 Dec  3 15:19 .
drwxr-xr-x 21 root root 4096 Dec  3 06:18 ..
drwxrwxrwt  2 root root 4096 May 21  2016 .ICE-unix
drwxrwxrwt  2 root root 4096 May 21  2016 .Test-unix
-r--r--r--  1 root root   11 Dec  3 14:45 .X0-lock
drwxrwxrwt  2 root root 4096 Dec  3 14:45 .X11-unix
drwxrwxrwt  2 root root 4096 May 21  2016 .XIM-unix
drwxrwxrwt  2 root root 4096 May 21  2016 .font-unix
drwxrwxrwt  4 xrdp xrdp 4096 Dec  3 14:45 .xrdp
drwx------  2 root root 4096 Dec  3 15:19 gadget-iNvIJk
drwx------  3 root root 4096 May 21  2016 systemd-private-c58ea7835c894b5e9e8e56250bd0dc6d-systemd-timesyncd.service-q5BdPO

root@beaglebone:~# mount -t gadgetfs none /tmp/gadget-iNvIJk
mount: unknown filesystem type 'gadgetfs'
root@beaglebone:~#

Ramanathan-R, 2016-12-03:
Closed by mistake in my last comment. Issue still open.

marcosox, 2016-12-03:
rmmod the modules g_multi and libcomposite as they prevent the USB host port from being used by the software.

Ramanathan-R, 2016-12-03:
Thank you, that did it. "usb-mitm" loads properly now.

Time to read up some more on the USB details :-)

closed 2016-12-03 by Ramanathan-R