usb-tools/USBProxy-legacy #64: Does it wotk

Andymion, 2017-07-04:


Andymion, 2017-07-04:
I use Armbian with mainline kernel. Gadgetfs is loaded (according to lsmod listing).
Module                  Size  Used by
uas                    20480  0
gadgetfs               24576  0
sun8i_codec_analog     24576  0
snd_soc_core          122880  1 sun8i_codec_analog
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_pcm                77824  2 snd_pcm_dmaengine,snd_soc_core
sun8i_ths              16384  0
thermal_sys            57344  1 sun8i_ths
uio_pdrv_genirq        16384  0
uio                    16384  1 uio_pdrv_genirq
usb_f_acm              16384  1
u_serial               20480  4 usb_f_acm
g_serial               16384  0
libcomposite           40960  2 g_serial,usb_f_acm

However I've got an error connected to gadgetfs:

Made directory /tmp/gadget-JddHVl for gadget
Error mounting gadgetfs from [/tmp/gadget-JddHVl].
Error code from mount is: [No such device]
searching in [/tmp/gadget-0kQbeB]
/tmp/gadget-0kQbeB device file not found.
Error, unable to find gadget file.
Fail on open 11 Resource temporarily unavailable
GadgetFS not connected.

Where should I dig in?

marcosox, 2017-07-04:
you need to rmmod also `libcomposite`, `g_serial` and `usb_f_acm`.
I agree that it should be documented somewhere, since this question comes up often (and I learned these things by trial and error).
Note that the modules must be removed **before** loading gadgetfs

Andymion, 2017-07-04:
Thanks a lot, marcosox! This was really unclear, but it works.

closed 2017-07-04 by Andymion