usb-tools/USBProxy-legacy #65: Segfault on init with "complicated" device

itdaniher, 2017-07-21:
```Program received signal SIGSEGV, Segmentation fault.
0xb6fbff58 in InterfaceGroup::add_interface(Interface*) () from /usr/local/lib/libUSBProxy.so.0
```

```
#0  0xb6fbff58 in InterfaceGroup::add_interface(Interface*) () from /usr/local/lib/libUSBProxy.so.0
#1  0xb6fbde18 in Configuration::add_interface(Interface*) () from /usr/local/lib/libUSBProxy.so.0
#2  0xb6fbdab2 in Configuration::Configuration(Device*, DeviceProxy*, int, bool) () from /usr/local/lib/libUSBProxy.so.0
#3  0xb6fbe508 in Device::Device(DeviceProxy*) () from /usr/local/lib/libUSBProxy.so.0
#4  0xb6fb450a in Manager::start_control_relaying() () from /usr/local/lib/libUSBProxy.so.0
#5  0x00012d52 in main ()
```

```Linux bananapi 4.11.5-sunxi #3 SMP Thu Jun 15 02:11:01 CEST 2017 armv7l GNU/Linux```
lsusb -vv output at [http://sprunge.us/deAQ](http://sprunge.us/deAQ)

itdaniher, 2017-07-21:
usbproxy -dd output prior to segfault at https://gist.github.com/itdaniher/0e6096c5bc3ea1baa80c3988d4fcc343

dominicgs, 2017-07-21:
I think this is almost certainly related to the 4 interfaces that the device has.  I'm not sure if we can support that, but it's where I'll start looking.

straithe, 2021-11-13:
I'm sorry this issue has not received a response in a while. @itdaniher are you still experiencing this issue?

closed 2021-12-12 by straithe