usb-tools/USBProxy-legacy #66: USBProxy crash on Huawei K3570

dpeddi, 2017-09-29:
Hi, 
I'm trying to analize the stream between my host and this device, but after starting usb-mitm, it produce a lot of errors and finally exit with abort. The log is really long to be attached, but probably the issue is on start.. so i would procede step by step.

I've added some printf and I can see that opening and writing complete correctly on the first interface while the other one fails. 

`
searching in [/tmp/gadget-yDT3DF]
Starting setup reader thread (7741) for EP00.
Starting setup writer thread (7742) for EP00.
Processing interface 0
Processing interface 0 alt:0
Processing interface 0 81
Opened EP81
Processing interface 0 82
Opened EP82
Processing interface 0 1
Opened EP01
Processing interface 1
Processing interface 1 alt:0
Processing interface 1 83
Error writing to EP 0x83 131 Invalid argument rc=-1
Opened EP83
Processing interface 2
Processing interface 2 alt:0
Processing interface 2 84
Error writing to EP 0x84 132 Invalid argument rc=-1
Opened EP84
[...]
`
Just for confirmation I've changed
`for (ifc_idx=0;ifc_idx<ifc_count;ifc_idx++) {`
to:
`for (ifc_idx=1;ifc_idx<ifc_count;ifc_idx++) {`

`
Processing interface 1
Processing interface 1 alt:0
Processing interface 1 83
Openingo EP 0x83 131 musb-hdrc
open_endpoint() opening path /tmp/gadget-8fgimi/ep3in
Error writing to EP 0x83 131 Invalid argument rc=-1 musb-hdrc
Opened EP83
Processing interface 2
Processing interface 2 alt:0
`
But opening on interface >1 still doesn't work..

`
lsusb -vv

Bus 001 Device 003: ID 12d1:1465 Huawei Technologies Co., Ltd..
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x12d1 Huawei Technologies Co., Ltd.
  idProduct          0x1465.
  bcdDevice            0.00
  iManufacturer           4 HUAWEI Technology
  iProduct                3 HUAWEI Mobile
  iSerial                 0.
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          201
[... 224 lines cut ...]
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x88  EP 8 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0001
  Self Powered
`

dpeddi, 2017-09-29:
with debug activated:

```
Opened EP01
Processing interface 1
Processing interface 1 alt:0
Processing interface 1 83
Opening EP 0x83 131 musb-hdrc
open_endpoint() opening path /tmp/gadget-MiQvhe/ep3in
EP83 buff: 01000000070583
Device: 12 01 00 02 ef 02 01 40 d1 12 65 14 00 00 04 03 00 01
  Manufacturer: HUAWEI Technology
  Product:      HUAWEI Mobile
<------>*Config(1): 09 02 c9 00 07 01 02 e0 fa
<------>   Name: Huawei   Configuration
<------><------>Interface(0):
<------><------><------>*Alt(0): 09 04 00 00 03 ff ff ff 00
<------><------><------>Other(0b): 08 0b 01 02 02 00 00 00
<------><------><------><------>EP(81): 07 05 81 03 40 00 05
<------><------><------><------>EP(82): 07 05 82 02 00 02 20
[...]
<------><------>Interface(6):
<------><------><------>*Alt(0): 09 04 06 00 02 08 06 50 00
<------><------><------><------>EP(06): 07 05 06 02 40 00 00
<------><------><------><------>EP(88): 07 05 88 02 40 00 00
Error writing to EP 0x83
Error writing to EP 0x83 131 Invalid argument rc=-1 musb-hdrc
Opened EP83
```
It seems that the write failure start happening after display->print();

dpeddi, 2017-09-29:
interesting reading:
https://sourceforge.net/p/libusb/mailman/message/35965754/

dpeddi, 2017-09-29:
Trying now with your Debian-USBProxy.img.xz but I get same issue...

S4mw1s3, 2017-10-13:
> interesting reading:
> https://sourceforge.net/p/libusb/mailman/message/35965754/

That is indeed interesting. I'm also experiencing these errno=22 errors with a simple usb<->rs232 device :(

```
libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 64
libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
libusb: debug [handle_events] poll() returned 1
libusb: debug [reap_for_handle] urb type=3 status=0 transferred=2
libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
libusb: debug [usbi_handle_transfer_completion] transfer 0x22fe600 has callback 0xb6bbfbd9
libusb: debug [sync_transfer_cb] actual_length=2
libusb: debug [libusb_free_transfer] transfer 0x22fe600
libusb: debug [libusb_alloc_transfer] transfer 0x22fe600
libusb: debug [libusb_submit_transfer] transfer 0x22fe600
libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 64
gadgetfs: 1 events received
Sending ACK
LibUSB> 40 09 10 00 00 00 00 00
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=22
libusb: debug [submit_bulk_transfer] first URB failed, easy peasy
libusb: debug [libusb_free_transfer] transfer 0x22fe600
Transfer error receiving on EP81 (xfertype 2): Input/Output Error
```

Which board and kernel are you using?
```
debian@beaglebone:~$ cat /proc/device-tree/model 
TI AM335x BeagleBone Black
debian@beaglebone:~$ 
debian@beaglebone:~$ uname -a
Linux beaglebone 4.9.52-ti-r64 #1 SMP PREEMPT Sat Sep 30 00:11:40 UTC 2017 armv7l GNU/Linux
debian@beaglebone:~$ 
```

dpeddi, 2017-10-13:
cat /proc/device-tree/model 
TI AM335x BeagleBone Black
root@beaglebone:~# 
root@beaglebone:~#  uname -a
Linux beaglebone 4.4.88-ti-r125 #1 SMP Thu Sep 21 19:23:24 UTC 2017 armv7l GNU/Linux
root@beaglebone:~# ^C

dpeddi, 2017-10-13:
thank you... probably I have another issue as well:
[  369.615003] musb_g_ep0_irq 804: SETUP packet len 0 != 8 ?

this seems gadgetfs related

S4mw1s3, 2017-10-14:
```
dpeddi  [23:15:44] 15 Transmit and 15 Receive Endpoints other than the mandatory Control Endpoint 0.
dpeddi  [23:17:24] lsusb -vv | grep -i endpointaddress| grep IN | wc -l 9
dpeddi  [23:17:31] lsusb -vv | grep -i endpointaddress| grep OUT | wc -l 6
dpeddi  [23:28:46] i've upgraded to debian 9.1 
dpeddi  [23:28:57] and now in dmesg i get : [   90.309997] musb_g_ep0_irq 804: SETUP packet len 0 != 8 ? 
dpeddi  [23:37:21] a guy posted a patch "https://www.spinics.net/lists/linux-usb/msg97114.html"
dpeddi  [23:37:30] probably never applied to "http://elixir.free-electrons.com/linux/v4.4.88/source/drivers/usb/musb/musb_gadget_ep0.c"
dpeddi  [23:41:09] but since my setup packet is 0 there isn't so much to dump :-(
```
From what I saw on IRC, you already got to the point where you got "SETUP packet len 0 != 8". How come you didn't experience the errno=22 error then? You used an older version of USBProxy?

dpeddi, 2017-10-14:
No i haven't got... I wrong...
What i found produce just a dump and wont solve.

dpeddi, 2018-02-10:
I'm trying to skip loading into gadgetfs of some interfaces: for example i'm not interested in the usb_storage this device provide... where  i should look to avoid the notification of such interface?

tnx

straithe, 2021-11-13:
Sorry for not responding to this in a timely manner. @dpeddi are you still experiencing this issue?

dpeddi, 2021-11-13:
Currently i'm busy in other projects. I don't have any more an usbproxy compatible hardware.

straithe, 2021-11-13:
Ok. I'm sorry I couldn't get to this in time. I am going to close this issue at this time. If you do get usbproxy compatible hardware again and would like to explore this issue, just tag me!

closed 2021-11-13 by straithe