usb-tools/USBProxy-legacy #73: (partial) success with ODroid C2, debugging needed

floe, 2018-01-19:
Hello everyone,

just wanted to let you know that I've successfully used `usb-mitm` on an ODroid C2 running the standard 3.14.79-117 kernel (with gadgetfs/inode.c patch applied). Thanks for this awesome tool!

However, the whole thing seems to be extremely fragile: I've only been able to successfully proxy one single device so far (an Arduino Micro operating as cdc-acm device).

Apart from that, I've tried four different HID devices, an audio device, and a mass storage device. Except for the mass storage device, none of the other ones would even enumerate on the host, all I got was

```
usb 3-3: new high-speed USB device number 8 using xhci_hcd
usb 3-3: device descriptor read/64, error 18
usb 3-3: device descriptor read/64, error 18
usb 3-3: new high-speed USB device number 9 using xhci_hcd
usb 3-3: device descriptor read/8, error -61
usb 3-3: device descriptor read/8, error -61
usb usb3-port3: unable to enumerate USB device
```

Note that the host thinks it's a high-speed device, even though it is in reality only full-speed.
I'm attaching the `lsusb -v` for the Arduino (which works) and one of the HID devices (which doesn't).

Any hints about how to start debugging this would be welcome.

Best, Florian
[arduino_desc.txt](https://github.com/dominicgs/USBProxy/files/1647947/arduino_desc.txt)
[logitech_desc.txt](https://github.com/dominicgs/USBProxy/files/1647948/logitech_desc.txt)

floe, 2018-01-19:
Just realized something: is it possible that the ODroid always enumerates as high-speed on a hardware level, and if the original device doesn't supply the secondary set of highspeed descriptors, then the enumeration will fail?

floe, 2018-01-22:
Update 2: after some aimless debugging, it seems that the gadgetfs stack on the ODroid chokes on device descriptors which don't have any OUT (host -> device) endpoints. When I add a bogus OUT endpoint, it suddenly starts working...

Lejurn, 2019-02-21:
Hi,
I have a similar issue regarding USBProxy and my odroid c2 (kernel version 3.16.62).
Storage devices are working fine, but HID devices don't.

USBProxy is stuck within the setup reader and writer loop. gadgetfs returns with the event code 4, GADGETFS_SUSPEND, I think.

`Starting setup reader thread (1024) for EP00.
Starting setup writer thread (1025) for EP00.
gadgetfs: 1 events received
gadgetfs: event 4
gadgetfs: 1 events received
gadgetfs: event 1
gadgetfs: 1 events received
gadgetfs: event 4
gadgetfs: 1 events received
gadgetfs: event 4
^CReceived SIGINT, stopping relaying...
`
I connected the odroid to my laptop and getting the following error messages within dmesg:

`[14202.882557] usb 1-4: new high-speed USB device number 72 using xhci_hcd
[14203.032381] usb 1-4: unable to read config index 0 descriptor/start: -61
[14203.032393] usb 1-4: can't read configurations, error -61
[14203.158574] usb 1-4: new high-speed USB device number 73 using xhci_hcd
[14203.307721] usb 1-4: unable to read config index 0 descriptor/start: -61
[14203.307732] usb 1-4: can't read configurations, error -61
[14203.307829] usb usb1-port4: attempt power cycle
[14203.958577] usb 1-4: new high-speed USB device number 74 using xhci_hcd
[14203.980788] usb 1-4: unable to read config index 0 descriptor/start: -61
[14203.980801] usb 1-4: can't read configurations, error -61
[14204.106572] usb 1-4: new high-speed USB device number 75 using xhci_hcd
[14204.128493] usb 1-4: unable to read config index 0 descriptor/start: -61
[14204.128505] usb 1-4: can't read configurations, error -61
[14204.128612] usb usb1-port4: unable to enumerate USB device
`
@floe  How did you add the bogus OUT endpoint?

Thank you very much.

floe, 2019-02-24:
I will see if I can dig up my modified sources, honestly cannot remember right now :-/

straithe, 2021-11-13:
Thank you for the update @floe ! I'm going to close this issue as there haven't been any updates to it in a while.

closed 2021-11-13 by straithe