usb-tools/USBProxy-legacy #74: FYI: Link added to Wireshark Wiki

hzulla, 2018-05-18:
Hi there,

thanks for your project. It took me a very long time to actually find it, despite googling various iterations of my problem.

I didn't end up using USBProxy, but it helped me find its cousin [SerialUSB](https://github.com/matlo/serialusb) which ended being the right tool for my problem.

To make things easier for others, I have asked to add the following text to the [Wireshark Wiki](https://wiki.wireshark.org/CaptureSetup/USB). Please let me know if you agree with this description of your project. Also, as I didn't actually use USBProxy, please let me know if the description is correct.

> MITM hardware with Linux
>
> If the USB host is a black-box device such as a game console and you cannot capture USB traffic on the host's operating system, here are two diy-projects that help you build a simple MITM device to sniff USB messages on the USB cable.
> 
> *SerialUSB* is designed to intercept USB HID traffic. Originally made for the GIMX project (which lets you connect PC game controllers to the PS4 by converting the HID protocol messages). You will need a Linux computer to capture the HID messages and an Arduino-based USB dongle. Parts are cheap. If you don't like soldering, you can buy ready-made "GIMX USB adapters" from the developer and from enthusiasts on eBay and elsewhere. 
> 
> *USBProxy* intercepts USB traffic with a Beaglebone Black, which is reconfigured to act as a USB gadget emulating the device connected to the 2nd USB port. According to the documentation, it allows sniffing non-HID traffic, as well.

dominicgs, 2018-05-19:
Looks good to me

hzulla, 2018-05-21:
Thank you. If you're unhappy with anything, I will gladly change the text for you. Thanks again for your project, it was very eye-opening and helpful.

closed 2018-05-21 by hzulla