usb-tools/USBProxy-legacy #75: Error, unable to find gadget file, GadgetFS not connected

jishnuvijay1991, 2018-05-29:
HI,
I am trying to enable usbProxy for Marvell PCB, and getting the following error while running usb-mitm

/tmp/gadget-e3CXsb device file not found.
Error, unable to find gadget file.
Fail on open 11 Resource temporarily unavailable
GadgetFS not connected.

l# usb-mitm -l
Loading plugins from /usr/lib/USBProxy/
vendorId=ffffffff
productId=ffffffff
cleaning up /tmp
removing 0
Made directory /tmp/gadget-e3CXsb for gadget
Printing Config data
        Strings: 2
                DeviceProxy: DeviceProxy_LibUSB
                HostProxy: HostProxy_GadgetFS
        Vectors: 1
                Plugins:
                        PacketFilter_StreamLog
Pointer: 1
                PacketFilter_StreamLog::file: 0x7f9b5ef3a8
Device: 12 01 00 03 ef 02 01 09 10 1e 00 33 00 00 01 02 03 01
  Manufacturer: Point Grey Research
  Product:      Grasshopper3 GS3-U3-28S5C
  Serial:       01141CD2
        *Config(1): 09 02 74 00 03 01 00 80 70
                Interface(0):
                        *Alt(0): 09 04 00 00 02 ef 05 00 00
                        Other(0b): 08 0b 00 03 ef 05 00 04
                        Other(24): 14 24 01 00 00 01 00 00 00 01 00 05 06 07 08 09 0a 0b 0c 0c
                        Other(30): 06 30 00 00 00 00
                        Other(30): 06 30 00 00 00 00
                                EP(02): 07 05 02 02 00 04 00
                                EP(82): 07 05 82 02 00 04 00
                Interface(1):
                        *Alt(0): 09 04 01 00 01 ef 05 01 00
                        Other(30): 06 30 00 00 00 00
                                EP(83): 07 05 83 02 00 04 00
                Interface(2):
                        *Alt(0): 09 04 02 00 01 ef 05 02 00
                        Other(30): 06 30 0f 00 00 00
                                EP(81): 07 05 81 02 00 04 00
HS Qualifier: 0a 06 10 02 00 00 00 40 01 00
         Config(2): 09 07 12 00 01 02 00 80 fa
                Interface(0):
                        *Alt(0): 09 04 00 00 00 ff 00 00 00
searching in [/tmp/gadget-e3CXsb]
/tmp/gadget-e3CXsb device file not found.
Error, unable to find gadget file.
Fail on open 11 Resource temporarily unavailable
GadgetFS not connected.
done

I had loaded both musb driver and gadgetFs

# lsmod 
Module                  Size  Used by
musb_hdrc              73027  0
gadgetfs               18305  1

I also tried the patch given in doc directory.
Thanks in advance...

marcosox, 2018-05-29:
I'll give it a try: did you also disable the other conflicting kernel modules? see #62

jishnuvijay1991, 2018-05-29:
Yes, I havn't loaded modules g_multi and libcomposite.

dominicgs, 2018-05-29:
@jishnuvijay1991 Could you mount gadgetfs somewhere and see what the file(s) it has available?  It sounds like it's not supported by our current list.

jishnuvijay1991, 2018-05-30:
It contains a sigle empty file "mvebu-u3d"

dominicgs, 2018-05-30:
That's not one of our supported platforms, but you could try adding it here: https://github.com/dominicgs/USBProxy/blob/master/src/Plugins/Hosts/GadgetFS_helpers.c#L188

jishnuvijay1991, 2018-05-30:
We had added the it in the array already..

jishnuvijay1991, 2018-05-30:
You are right,
Issue is resolved.

closed 2018-05-30 by jishnuvijay1991