usb-tools/USBProxy-legacy #76: Can't get USBProxy to work on Raspberry Pi Zero W

mweal-ed, 2018-12-17:
I have been try to get USBProxy to work on a Raspberry Pi Zero W without a lot of success.  I am using the on board USB for the gadget port and a MAX3421E for the host port (similar to Raspdancer 21).  Do you have any suggestion on what I could be doing wrong?

I use the following to build USBProxy:

~~~
cd
sudo apt-get install build-essential cmake libpcap-dev libusb-1.0-0-dev
wget https://github.com/dominicgs/USBProxy/archive/master.zip 
unzip master.zip
cd USBProxy-master
mkdir src/build
cd src/build
cmake ..
make

sudo make install
sudo ldconfig
~~~

* add "dtoverlay=dwc2" to /boot/config.txt
* add " modules-load=dwc2 " after rootwait string in /boot/cmdline.txt.

* to prevent usb printer drivers from grabbing usb port
~~~
sudo sh -c 'echo "blacklist usblp" > /etc/modprobe.d/blacklist-usblp.conf
~~~

~~~
sudo reboot
sudo  usb-mitm -l
~~~

I get the following output
~~~

Loading plugins from /usr/local/lib/USBProxy/
vendorId=ffffffff
productId=ffffffff
cleaning up /tmp
removing 1
Made directory /tmp/gadget-8rFsub for gadget
Printing Config data
        Strings: 2
                DeviceProxy: DeviceProxy_LibUSB
                HostProxy: HostProxy_GadgetFS
        Vectors: 1
                Plugins:
                        PacketFilter_StreamLog
Pointer: 1
                PacketFilter_StreamLog::file: 0xb6c8fcb0
Device: 12 01 00 02 00 00 00 40 b8 04 02 02 00 01 01 02 03 01
  Manufacturer: EPSON
  Product:      TM-T88V
  Serial:       4D5844469389310000
        *Config(1): 09 02 20 00 01 01 00 e0 01
                Interface(0):
                        *Alt(0): 09 04 00 00 02 ff ff 02 00
                                EP(01): 07 05 01 02 40 00 00
                                EP(82): 07 05 82 02 40 00 00
old bInterval: 00
calculated new bInterval: 80000000
old bInterval: 00
calculated new bInterval: 80000000
searching in [/tmp/gadget-8rFsub]
/tmp/gadget-8rFsub device file not found.
Error, unable to find gadget file.
Fail on open 34 Numerical result out of range
GadgetFS not connected.
done
~~~

mweal-ed, 2018-12-18:
I have not done any serious testing yet, but adding "20980000.usb", to  *devices[]  in src/Plugins/Hosts/GadgetFS_helpers.c  find_gadget_filename() seems to solve the problem.

Trakt0r22, 2020-07-22:
I cannot make it work in my Pi Zero W either, spite following the advice from @mweal-ed, adding "20980000.usb" to the devices[] in find_gadget_filename(). Im running a P4WNP1-aloa img (based on Kali Linux) and the errors I get when I run "usb-mitm -k" is:

`Loading plugins from /usr/local/lib/USBProxy/
vendorId=ffffffff
productId=ffffffff
cleaning up /tmp
removing 1
Made directory /tmp/gadget-LSeMna for gadget
Error mounting gadgetfs from [/tmp/gadget-LSeMna].
Error code from mount is: [No such device]
Printing Config data
	Strings: 2
		DeviceProxy: DeviceProxy_LibUSB
		HostProxy: HostProxy_GadgetFS
	Vectors: 1
		Plugins:
			PacketFilter_KeyLogger
Pointer: 1
		PacketFilter_KeyLogger::file: 0xb6cdfcf8
Unable to connect to device proxy.
done`

This is what my install progress looked like:

`root@kali:~# cmake .. && make && sudo make install && ldconfig
-- Could NOT find LORCON (missing: LORCON_LIBRARIES LORCON_INCLUDE_DIR) 
-- Pcap include dirs set to /usr/include
-- Pcap library set to  /usr/lib/arm-linux-gnueabi/libpcap.so
-- Configuring done
-- Generating done
-- Build files have been written to: /root/USBProxy-legacy/src/build
[ 36%] Built target USBProxy
[ 39%] Built target DeviceProxy_LibUSB
[ 43%] Built target DeviceProxy_Callback
[ 46%] Built target DeviceProxy_TCP
[ 50%] Built target DeviceProxy_Loopback
Scanning dependencies of target HostProxy_GadgetFS
[ 51%] Building C object Plugins/Hosts/CMakeFiles/HostProxy_GadgetFS.dir/GadgetFS_helpers.c.o
/root/USBProxy-legacy/src/Plugins/Hosts/GadgetFS_helpers.c: In function 'clean_tmp':
/root/USBProxy-legacy/src/Plugins/Hosts/GadgetFS_helpers.c:65:3: warning: 'readdir_r' is deprecated [-Wdeprecated-declarations]
   65 |   if (readdir_r(dir, entry, &result)) break;
      |   ^~
In file included from /root/USBProxy-legacy/src/Plugins/Hosts/GadgetFS_helpers.c:31:
/usr/include/dirent.h:183:12: note: declared here
  183 | extern int readdir_r (DIR *__restrict __dirp,
      |            ^~~~~~~~~
/root/USBProxy-legacy/src/Plugins/Hosts/GadgetFS_helpers.c: In function 'find_gadget_filename':
/root/USBProxy-legacy/src/Plugins/Hosts/GadgetFS_helpers.c:230:3: warning: 'readdir_r' is deprecated [-Wdeprecated-declarations]
  230 |   if (readdir_r(dir, entry, &result)) break;
      |   ^~
In file included from /root/USBProxy-legacy/src/Plugins/Hosts/GadgetFS_helpers.c:31:
/usr/include/dirent.h:183:12: note: declared here
  183 | extern int readdir_r (DIR *__restrict __dirp,
      |            ^~~~~~~~~
[ 53%] Linking CXX shared library HostProxy_GadgetFS.so
[ 55%] Built target HostProxy_GadgetFS
[ 58%] Built target HostProxy_TCP
[ 62%] Built target PacketFilter_KeyLogger
[ 65%] Built target PacketFilter_Callback
[ 68%] Built target PacketFilter_ROT13
[ 72%] Built target PacketFilter_UDPHID
[ 75%] Built target PacketFilter_MassStorage
[ 79%] Built target PacketFilter_Power
[ 82%] Built target PacketFilter_StreamLog
[ 86%] Built target PacketFilter_PcapLogger
[ 89%] Built target Injector_UDPHID
[ 93%] Built target Injector_UDP
[ 93%] Built target version.h
[ 96%] Built target usb-mitm
[100%] Built target USBProxyAPI
[... 45 lines cut ...]
-- Up-to-date: /usr/local/include/USBProxy/USBString.h
-- Up-to-date: /usr/local/lib/pkgconfig/usbproxy.pc
-- Up-to-date: /usr/local/lib/USBProxy/DeviceProxy_Loopback.so
-- Up-to-date: /usr/local/lib/USBProxy/DeviceProxy_TCP.so
-- Up-to-date: /usr/local/lib/USBProxy/DeviceProxy_Callback.so
-- Up-to-date: /usr/local/lib/USBProxy/DeviceProxy_LibUSB.so
-- Up-to-date: /usr/local/lib/USBProxy/HostProxy_TCP.so
-- Installing: /usr/local/lib/USBProxy/HostProxy_GadgetFS.so
-- Set runtime path of "/usr/local/lib/USBProxy/HostProxy_GadgetFS.so" to ""
-- Up-to-date: /usr/local/lib/USBProxy/PacketFilter_Callback.so
-- Up-to-date: /usr/local/lib/USBProxy/PacketFilter_KeyLogger.so
-- Up-to-date: /usr/local/lib/USBProxy/PacketFilter_ROT13.so
-- Up-to-date: /usr/local/lib/USBProxy/PacketFilter_UDPHID.so
-- Up-to-date: /usr/local/lib/USBProxy/PacketFilter_MassStorage.so
-- Up-to-date: /usr/local/lib/USBProxy/PacketFilter_Power.so
-- Up-to-date: /usr/local/lib/USBProxy/PacketFilter_StreamLog.so
-- Up-to-date: /usr/local/lib/USBProxy/PacketFilter_PcapLogger.so
-- Up-to-date: /usr/local/lib/USBProxy/Injector_UDP.so
-- Up-to-date: /usr/local/lib/USBProxy/Injector_UDPHID.so
-- Up-to-date: /usr/local/bin/usb-mitm
-- Up-to-date: /usr/local/lib/libUSBProxyAPI.so.0.1
-- Up-to-date: /usr/local/lib/libUSBProxyAPI.so.0
-- Up-to-date: /usr/local/lib/libUSBProxyAPI.so`

Any advice/help?

foxmiti, 2020-12-17:
add dtoverlay=dwc2, dr_mode=peripheral in config.txt to enable device mode on UDC for rpi zero w.

straithe, 2021-11-13:
I'm sorry no one has responded to this issue in a while @mweal-ed . It looks like @foxmiti has provided a suitable answer so I am going to close this issue at this time. Please re-open this issue or open a new one if assistance is still needed.

closed 2021-11-13 by straithe

mweal-ed, 2021-11-13:
actually if look above I had solved this problem by adding  "20980000.usb", to *devices[] in src/Plugins/Hosts/GadgetFS_helpers.c

straithe, 2021-11-13:
Perfect! Thank you for that clarification.