usb-tools/USBProxy-legacy #78: USBProxy on Kernel 4.14.71-ti-r80

MrSliphe, 2019-02-07:
Hello,
I'm currently trying to get USBProxy running on the BeagleBone Black using the latest image release from beagleboard.org. The kernel version of this is 4.14.71-ti-r80. I'm using this
https://github.com/RobertCNelson/ti-linux-kernel-dev/releases/tag/4.14.71-ti-r80
to integrate the patches, but since the source code has changed, I tried doing it by hand. As far as I understand, I would just need to apply the two changes in the inode.c.patch. Rebuilding using the rebuild.sh also works (at least it throws no errors).

Using USBProxy on the Beaglebone though makes me think, that the patches are not applied correctly, as it throws this error when using it:
```
Loading plugins from /usr/local/lib/USBProxy/
vendorId=ffffffff
productId=ffffffff
cleaning up /tmp
removing 1
Made directory /tmp/gadget-dzayif for gadget
Error mounting gadgetfs from [/tmp/gadget-dzayif].
Error code from mount is: [No such device]
Printing Config data
	Strings: 2
		DeviceProxy: DeviceProxy_LibUSB
		HostProxy: HostProxy_GadgetFS
	Vectors: 0
Pointer: 0
Device: 12 01 10 01 00 00 00 08 8a 25 01 00 00 02 01 02 00 01
  Manufacturer: SINO WEALTH
  Product:      USB KEYBOARD
	*Config(1): 09 02 3b 00 02 01 00 a0 32
		Interface(0):
			*Alt(0): 09 04 00 00 01 03 01 01 00
				HID: 09 21 10 01 00 01 22 41 00
				EP(81): 07 05 81 03 08 00 0a
		Interface(1):
			*Alt(0): 09 04 01 00 01 03 00 00 00
				HID: 09 21 10 01 00 01 22 9e 00
				EP(82): 07 05 82 03 08 00 0a
old bInterval: 0A
calculated new bInterval: 07
old bInterval: 0A
calculated new bInterval: 07
searching in [/tmp/gadget-dzayif]
/tmp/gadget-dzayif device file not found.
Error, unable to find gadget file.
Fail on open 11 Resource temporarily unavailable
GadgetFS not connected.
done
```

`lsmod | grep "gadgetfs"` shows the module is loaded, so the problem must be with how I applied the patch though I'm not sure what I did wrong. I just looked for the corresponding part in the inode.c (they are approx. 100 lines further up than the inode.c.patch tries to insert them) and inserted the spin_unlock_irq and spin_lock_irq lines. The second block appeared to be already applied in the newer inode.c so I only added the first block around line 559.

I tested using the same USB-device with the image provided in releases and there it worked. Is it possible to use USBProxy with this image or any image using a 4.X kernel or is it not compatible?

mweal-ed, 2019-02-08:
I guessing here as I had the similar error message with the pi zero ( issue #76 ).  Is it possible they changed the filename for the gadgetfs usb in the sys tree?

Here is what I did: 
* check for a file in the format *.usb in the /sys/devices/platform/soc/ (or is it /sys/devices/platform/ocp/ on the beagbone). (or print  entry->d_name in the while loop of src/Plugins/Hosts/GadgetFS_helpers.c find_gadget_filename())
* check if this file name is in  *devices[] in src/Plugins/Hosts/GadgetFS_helpers.c find_gadget_filename().  
* If not add it and recompile/install.

hope this helps

MrSliphe, 2019-02-27:
Sorry, I just had time to test your tips. Sadly it did not work.
I checked in both /sys/devices/platform/soc/ and /sys/devices/platform/ocp/ and found the folder 47400000.usb in ocp. Adding it to *devices[] in find_gadget_filename() just before the NULL did not help. The error is still the same after recompiling. In the 47400000.usb folder there are more folders with *.usb maybe adding those would help? I will try later.

Regarding the print of entry->d_name: When I added a fprintf line to print entry-d_name I could see, that the while loop is looped through 3 times. The first time the print was empty, the second time it was .. and the third time . (so just a dot). Maybe this is an indicator for something else?

Thanks regardless.

mweal-ed, 2019-02-27:
I'm working on a pi zero w, and don't have access to a bbb right now, so I can't try it, but I get the following in the while loop on the Pi.  Sorry I can't be of any more help.
```
.
..
20980000.usb (your .usb will be different)
```

 "sudo usb-mitm -l" ?

MrSliphe, 2019-02-27:
The empty line was me not thinking correctly, I added the print 
```
fprintf(stderr, "%s\n", entry->d_name);
```
before the
```
if (readdir_r(dir, entry, &result)) break;
```
Although I still do not see my .usb if I add it after that line. Using "usb-mitm -l" as root the output is now:
```
..
.
.
```

What did you do to apply the inode.c.patch? Just adjust the two parts manually and what did you change? I still hope it just has something to do with me patching wrong (and maybe adding the right part to devices[]).

mweal-ed, 2019-02-27:
I have not loaded the patch.  I originally thought it wasn't needed anymore.  You got me looking into it further and now I'm questioning that decision.

MrSliphe, 2019-02-27:
Okay I have now fixed it, the issue was not with the patch or USBProxy. Thus I did not need to add to the device list, as it was "musb-hdrc" for me, which is already included. The error came, because with the image I used, libcomposite was loaded at boot, which conflicts with the use of gadgetfs.
By editing the `use_libcomposite ()` in
```
opt/scripts/boot/am335x_evm.sh
```
and removing (or commenting) the following lines
```
echo "${log} modprobe libcomposite"
modprobe libcomposite || true
if [ -d /sys/module/libcomposite ] ; then
        run_libcomposite
else
        if [ -f /sbin/depmod ] ; then
                /sbin/depmod -a
        fi
        echo "${log} ERROR: [libcomposite didn't load]"
fi
```
the conflicting module wasn't loaded and I could freely use gadgetfs.

Thank you very much @mweal-ed for your help! Even if the solution was something else you gave me the idea that it may be something specific with the BeagleBone Black and gadgetfs.

closed 2019-02-27 by MrSliphe

palesius, 2019-02-28:
For what it's worth, (And it has been a LONG LONG time, since I have
touched the current source). If the inode patch was not working it would
totally lock up the device, I think it gets stuck indefinitely in the
gadgetfs driver. You would not just get a simple error message (unless
things have changed since then).

mweal-ed, 2019-02-28:
yes, @palesius  that agrees with what I found on the history of the patch at https://www.spinics.net/lists/linux-usb/msg99032.html