usb-tools/USBProxy-legacy #79: Memory Leak

mweal-ed, 2019-02-28:
I have USBProxy running on a Pi zero w and seem to be experiencing a memory leak.   

When running "sudo  usb-mitm -l" (with very little USB traffic)  I get a linux oom error after a couple minutes.  By running "free -m -s" in another window I see about 1MB per second of memory is being eaten. I have not been able to run this through valgrind as it does not support the Pi zero cpu, so debugging is going to be a bit more difficult.

1) I was wondering if  someone could check this on a BBB, so I know if it is a Pi issue or USBProxy Issue.  
2) Does Anyone suggestions as to what the problem may be.  
```
[87207.606922] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep da652e50 ep1out, 0)
[87207.608147] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep da652f10 ep2in, 0)
[87210.709040] usb-mitm invoked oom-killer: gfp_mask=0x14080c0(GFP_KERNEL|__GFP_ZERO), nodemask=(null),  order=0, oom_score_adj=0
[87210.709080] CPU: 0 PID: 15352 Comm: usb-mitm Tainted: G         C O    4.14.79+ #1159
[87210.709085] Hardware name: BCM2835
[87210.709142] [<c0016420>] (unwind_backtrace) from [<c0013d40>] (show_stack+0x20/0x24)
[87210.709171] [<c0013d40>] (show_stack) from [<c0634c44>] (dump_stack+0x20/0x28)
[87210.709203] [<c0634c44>] (dump_stack) from [<c00fac54>] (dump_header+0x94/0x1f4)
[87210.709223] [<c00fac54>] (dump_header) from [<c00fa0c0>] (oom_kill_process+0x3a0/0x4cc)
[87210.709241] [<c00fa0c0>] (oom_kill_process) from [<c00fa924>] (out_of_memory+0x10c/0x32c)
[87210.709259] [<c00fa924>] (out_of_memory) from [<c00ffe58>] (__alloc_pages_nodemask+0xda4/0xf10)
[87210.709286] [<c00ffe58>] (__alloc_pages_nodemask) from [<c0129158>] (__pte_alloc+0x30/0x1c8)
[87210.709305] [<c0129158>] (__pte_alloc) from [<c012c824>] (handle_mm_fault+0xaa4/0xb7c)
[87210.709327] [<c012c824>] (handle_mm_fault) from [<c064f4a8>] (do_page_fault+0x324/0x3b8)
[87210.709344] [<c064f4a8>] (do_page_fault) from [<c064f5e8>] (do_translation_fault+0xac/0xb4)
[87210.709360] [<c064f5e8>] (do_translation_fault) from [<c0009200>] (do_DataAbort+0x44/0xc4)
[87210.709374] [<c0009200>] (do_DataAbort) from [<c064ee24>] (__dabt_usr+0x44/0x60)
[87210.709381] Exception stack(0xd8c2bfb0 to 0xd8c2bff8)
[87210.709392] bfa0:                                     00000209 00000000 1d3fff00 0001bef9
[87210.709404] bfc0: b6c397c4 1d3fff08 b6c397c4 b6c397c4 1d400108 0000270f 000001ff b6c1d410
[87210.709413] bfe0: 00000209 b3efecd8 00000001 b6b6d5d0 60000010 ffffffff
[87210.709418] Mem-Info:
[87210.709440] active_anon:49870 inactive_anon:49905 isolated_anon:0
                active_file:20 inactive_file:28 isolated_file:0
                unevictable:0 dirty:0 writeback:0 unstable:0
                slab_reclaimable:1255 slab_unreclaimable:1508
                mapped:71 shmem:71 pagetables:503 bounce:0
                free:5668 free_pcp:146 free_cma:1705
[87210.709458] Node 0 active_anon:199480kB inactive_anon:199620kB active_file:80kB inactive_file:112kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:284kB dirty:0kB writeback:0kB shmem:284kB writeback_tmp:0kB unstable:0kB all_unreclaimable? no
[87210.709483] Normal free:22672kB min:16384kB low:20480kB high:24576kB active_anon:199480kB inactive_anon:199620kB active_file:80kB inactive_file:112kB unevictable:0kB writepending:0kB present:458752kB managed:443872kB mlocked:0kB kernel_stack:752kB pagetables:2012kB bounce:0kB free_pcp:584kB local_pcp:584kB free_cma:6820kB
[87210.709486] lowmem_reserve[]: 0 0
[87210.709498] Normal: 276*4kB (UMEC) 154*8kB (UMEC) 113*16kB (UMEC) 37*32kB (UMEC) 21*64kB (UMEC) 15*128kB (UEC) 3*256kB (UC) 6*512kB (UME) 2*1024kB (UE) 2*2048kB (UC) 1*4096kB (C) = 22672kB
[87210.709567] 161 total pagecache pages
[87210.709572] 38 pages in swap cache
[87210.709578] Swap cache stats: add 158493, delete 158455, find 8316/14614
[87210.709582] Free swap  = 0kB
[87210.709586] Total swap = 102396kB
[87210.709589] 114688 pages RAM
[87210.709592] 0 pages HighMem/MovableOnly
[87210.709597] 3720 pages reserved
[87210.709600] 2048 pages cma reserved
[87210.709604] [ pid ]   uid  tgid total_vm      rss nr_ptes nr_pmds swapents oom_score_adj name
[87210.709633] [   70]     0    70     6500      108      10       0       68             0 systemd-journal
[87210.709643] [  100]     0   100     3584       14       8       0      135         -1000 systemd-udevd
[87210.709657] [  184]   100   184     4316       19       9       0       94             0 systemd-timesyn
[87210.709667] [  208]     0   208     1843       26       7       0       79             0 systemd-logind
[87210.709678] [  212]     0   212     5745       54       8       0      133             0 rsyslogd
[87210.709688] [  213]   108   213     1597        5       7       0       79             0 avahi-daemon
[87210.709700] [  233]     0   233     1329       15       6       0       45             0 cron
[87210.709709] [  236]   108   236     1597        0       7       0       76             0 avahi-daemon
[87210.709722] [  238]   105   238     1626        1       7       0      101          -900 dbus-daemon
[87210.709731] [  246] 65534   246     1322        4       6       0       56             0 thd
[87210.709741] [  328]     0   328     2532       27       8       0      105             0 wpa_supplicant
[87210.709753] [  340]     0   340      520        0       4       0       33             0 hciattach
[87210.709766] [  351]     0   351     1767        0       7       0       75             0 bluetoothd
[87210.709778] [  414]     0   414      717       14       5       0       69             0 dhcpcd
[87210.709790] [  422]     0   422    12265       42      13       0      113             0 automount
[87210.709800] [  436]     0   436     2548        1       8       0      136         -1000 sshd
[87210.709811] [  438]     0   438     1045        0       6       0       34             0 agetty
[87210.709823] [  523]     0   523     2878       14       9       0      171             0 sshd
[87210.709834] [  528]  1000   528     2412        0       8       0      170             0 systemd
[87210.709844] [  531]  1000   531     7431        0      11       0      324             0 (sd-pam)
[87210.709855] [  538]  1000   538     2914       42       9       0      156             0 sshd
[87210.709865] [  541]  1000   541     1543       46       7       0      294             0 bash
[87210.709878] [  798]     0   798     2878       16       9       0      169             0 sshd
[87210.709888] [  808]  1000   808     2949       35       9       0      197             0 sshd
[87210.709899] [  811]  1000   811     1514        5       7       0      305             0 bash
[87210.709909] [ 7389]     0  7389     2878        0       9       0      185             0 sshd
[87210.709919] [ 7399]  1000  7399     2911        0       9       0      197             0 sshd
[87210.709931] [ 7402]  1000  7402     1557        0       7       0      354             0 bash
[87210.709944] [15033]     0 15033     1803        0       7       0       91             0 sudo
[87210.709954] [15037]     0 15037     1715        0       7       0       94             0 su
[87210.709964] [15042]     0 15042     1300        0       6       0       97             0 bash
[87210.709977] [15342]  1000 15342     1663       18       7       0       38             0 free
[87210.709989] [15344]     0 15344   134914    99198     245       0    19596             0 usb-mitm
[87210.709997] Out of memory: Kill process 15344 (usb-mitm) score 845 or sacrifice child
[87210.710072] Killed process 15344 (usb-mitm) total-vm:539656kB, anon-rss:396792kB, file-rss:0kB, shmem-rss:0kB
[87210.954812] oom_reaper: reaped process 15344 (usb-mitm), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB
[87210.969113] gadgetfs: disconnected
```

mweal-ed, 2019-02-28:
Managed to get my BBB back, so was able to verify the memory leak is on the BBB too, so this is not a Pi issue.
```
debian@arm:~$ sudo usb-mitm -l
[sudo] password for debian:
Loading plugins from /usr/local/lib/USBProxy/
vendorId=ffffffff
productId=ffffffff
cleaning up /tmp
removing 1
Made directory /tmp/gadget-LVdjYu for gadget
Printing Config data
        Strings: 2
                DeviceProxy: DeviceProxy_LibUSB
                HostProxy: HostProxy_GadgetFS
        Vectors: 1
                Plugins:
                        PacketFilter_StreamLog
Pointer: 1
                PacketFilter_StreamLog::file: 0xb6d55540
Device: 12 01 00 02 00 00 00 40 b8 04 02 02 00 01 01 02 03 01
  Manufacturer: EPSON
  Product:      TM-T88V
  Serial:       4D5844469389310000
        *Config(1): 09 02 20 00 01 01 00 e0 01
                Interface(0):
                        *Alt(0): 09 04 00 00 02 ff ff 02 00
                                EP(01): 07 05 01 02 40 00 00
                                EP(82): 07 05 82 02 40 00 00
old bInterval: 00
calculated new bInterval: 80000000
old bInterval: 00
calculated new bInterval: 80000000
searching in [/tmp/gadget-LVdjYu]
Starting setup writer thread (2149) for EP00.
Starting setup reader thread (2148) for EP00.
[80 06 03 03 09 04 ff 00]
[80 06 03 03 09 04 26 00]:
        26 03 34 00 44 00 35 00 38 00 34 00 34 00 34 00 36 00 39 00 33 00 38 00                                                                               39 00 33 00 31 00 30 00
        30 00 30 00 30 00
[80 06 00 03 00 00 ff 00]
[80 06 00 03 00 00 04 00]: 04 03 09 04
[80 06 02 03 09 04 ff 00]
[80 06 02 03 09 04 10 00]: 10 03 54 00 4d 00 2d 00 54 00 38 00 38 00 56 00
[80 06 02 03 00 00 00 04]
[80 06 02 03 00 00 10 00]: 10 03 54 00 4d 00 2d 00 54 00 38 00 38 00 56 00
[00 09 01 00 00 00 00 00]
Opened EP01
Opened EP82
[c1 00 00 00 00 00 02 00]
[c1 00 00 00 00 00 02 00]: 26 30
Starting writer thread (2154) for EP82.
[41 06 00 30 00 00 00 00]
[c1 00 00 00 00 00 02 00]
Starting reader thread (2153) for EP82.
Starting writer thread (2152) for EP01.
[c1 00 00 00 00 00 02 00]: 26 30
Starting reader thread (2151) for EP01.
[80 06 00 03 00 00 00 04]
[80 06 00 03 00 00 04 00]: 04 03 09 04
[80 06 01 03 09 04 00 04]
[80 06 01 03 09 04 0c 00]: 0c 03 45 00 50 00 53 00 4f 00 4e 00
[80 06 02 03 09 04 00 04]
[80 06 02 03 09 04 10 00]: 10 03 54 00 4d 00 2d 00 54 00 38 00 38 00 56 00
[80 06 03 03 09 04 00 04]
[80 06 03 03 09 04 26 00]:
        26 03 34 00 44 00 35 00 38 00 34 00 34 00 34 00 36 00 39 00 33 00 38 00                                                                               39 00 33 00 31 00 30 00
        30 00 30 00 30 00
[c1 00 00 00 00 00 02 00]
[c1 00 00 00 00 00 02 00]: 26 30
[c1 01 00 00 00 00 02 00]
[... 144 lines cut ...]
[  563.117005] [ 1923]   101  1923      647       66       4        0             0 dbus-daemon
[  563.125894] [ 1950]   102  1950      716       83       4        0             0 avahi-daemon
[  563.134870] [ 1951]   102  1951      691       63       4        0             0 avahi-daemon
[  563.143843] [ 2080]     0  2080     1298      114       4        0         -1000 sshd
[  563.152086] [ 2104]     0  2104      836       42       5        0             0 getty
[  563.160420] [ 2105]     0  2105      836       42       4        0             0 getty
[  563.168755] [ 2106]     0  2106      836       42       4        0             0 getty
[  563.177090] [ 2107]     0  2107      836       42       4        0             0 getty
[  563.185427] [ 2108]     0  2108      836       42       4        0             0 getty
[  563.193762] [ 2109]     0  2109      836       42       4        0             0 getty
[  563.202097] [ 2110]     0  2110      416       41       4        0             0 getty
[  563.210432] [ 2117]     0  2117     2074      168       6        0             0 sshd
[  563.218676] [ 2119]  1000  2119     2107      172       5        0             0 sshd
[  563.226918] [ 2120]  1000  2120     1032       94       5        0             0 bash
[  563.235162] [ 2127]     0  2127     2074      168       6        0             0 sshd
[  563.243404] [ 2129]  1000  2129     2107      172       5        0             0 sshd
[  563.251647] [ 2130]  1000  2130     1033       94       4        0             0 bash
[  563.259889] [ 2140]  1000  2140      747       31       4        0             0 free
[  563.268163] [ 2143]     0  2143     1018       84       4        0             0 sudo
[  563.276415] [ 2144]     0  2144   136591   121206     248        0             0 usb-mitm
[  563.285022] Out of memory: Kill process 2144 (usb-mitm) score 925 or sacrifice child
[  563.293176] Killed process 2144 (usb-mitm) total-vm:546364kB, anon-rss:484704kB, file-rss:120kB
[  563.539599] gadgetfs: disconnected

```

mweal-ed, 2019-03-01:
Looks like this is a duplicate of issue #6. 

I have tried running the code under valgrind on the BBB, but it does not work. 
I tried heapusage and all it tells me is I have a leak on the heap.
I was unable to run memleax on either the pi or the BBB.

mweal-ed, 2019-03-11:
Looks like the problem is that DeviceProxy_LibUSB::receive_data allocates memory for all received packet data.  Normally RelayReader::relay_read enqueue this buffer, and it is freed after the data in dequeued.   However when a zero length packet is received, the buffer is never enqueued and thus is never dequeued and freed.  The fix is  changing the following in RelayReader::relay_read from

```
		if (length)
			_sendQueue->enqueue(std::make_shared<Packet>(endpoint, buf, length));

```
to
```
		if (length)
			_sendQueue->enqueue(std::make_shared<Packet>(endpoint, buf, length));
		else
			free(buf);
```

straithe, 2021-11-13:
Hi, @mweal-ed are you still experiencing this issue?

mweal-ed, 2021-11-13:
It has been a while since I worked on this project, but the fix above had solved the memory leak problem.

straithe, 2021-11-13:
Thank you for the update. I'll close this issue.

closed 2021-11-13 by straithe