usb-tools/USBProxy-legacy #82: usb-mitm doesn't quit  gracefully on Ctrl+C

zapa1928, 2020-06-10:
Version ShmooCon_2015-109-g6a04
Running under kernel 4.19.94-ti-r42

When I try to exit the usm-mitm it tries to exit but doesn't finish. 

`debian@beaglebone:~$ sudo usb-mitm -ldd
[sudo] password for debian:
Version ShmooCon_2015-109-g6a04
Running under kernel 4.19.94-ti-r42
CP: String DeviceProxy = DeviceProxy_LibUSB
CP: String HostProxy = HostProxy_GadgetFS
Loading plugins from /usr/local/lib/USBProxy/
vendorId=ffffffff
productId=ffffffff
cleaning up /tmp
removing 1
Made directory /tmp/gadget-fTMRsk for gadget
Printing Config data
        Strings: 2
                DeviceProxy: DeviceProxy_LibUSB
                HostProxy: HostProxy_GadgetFS
        Vectors: 1
                Plugins:
                        PacketFilter_StreamLog
Pointer: 1
                PacketFilter_StreamLog::file: 0xb6d39d10
Connected to device: 067b:2303@02 Prolific Technology Inc. - USB-Serial Controller

LibUSB> 80 06 00 01 00 00 12 00
LibUSB< 12 01 01 01 00 00 00 40 7b 06 03 23 00 03 01 02 00 01
LibUSB> 80 06 00 03 00 00 08 00
LibUSB< 04 03 09 04
LibUSB> 80 06 01 03 09 04 08 00
LibUSB< 32 03 50 00 72 00 6f 00
LibUSB> 80 06 01 03 09 04 32 00
LibUSB<
        32 03 50 00 72 00 6f 00 6c 00 69 00 66 00 69 00 63 00 20 00 54 00 65 00 63 00 68 00 6e 00 6f 00
        6c 00 6f 00 67 00 79 00 20 00 49 00 6e 00 63 00 2e 00
LibUSB> 80 06 02 03 09 04 08 00
LibUSB< 2c 03 55 00 53 00 42 00
LibUSB> 80 06 02 03 09 04 2c 00
LibUSB<
        2c 03 55 00 53 00 42 00 2d 00 53 00 65 00 72 00 69 00 61 00 6c 00 20 00 43 00 6f 00 6e 00 74 00
        72 00 6f 00 6c 00 6c 00 65 00 72 00
LibUSB> 80 06 00 02 00 00 09 00
LibUSB< 09 02 27 00 01 01 00 a0 32
LibUSB> 80 06 00 02 00 00 27 00
LibUSB<
        09 02 27 00 01 01 00 a0 32 09 04 00 00 03 ff 00 00 00 07 05 82 03 20 00 01 07 05 01 02 20 00 00
        07 05 81 02 20 00 00
LibUSB> 80 06 00 06 00 00 0a 00
Error sending setup packet: Pipe error
LibUSB> 80 08 00 00 00 00 01 00
LibUSB< 01
Device: 12 01 01 01 00 00 00 40 7b 06 03 23 00 03 01 02 00 01
  Manufacturer: Prolific Technology Inc.
  Product:      USB-Serial Controller
        *Config(1): 09 02 27 00 01 01 00 a0 32
                Interface(0):
                        *Alt(0): 09 04 00 00 03 ff 00 00 00
                                EP(82): 07 05 82 03 20 00 01
                                EP(01): 07 05 01 02 20 00 00
                                EP(81): 07 05 81 02 20 00 00
old bInterval: 01
calculated new bInterval: 04
old bInterval: 00
calculated new bInterval: 80000000
old bInterval: 00
calculated new bInterval: 80000000

        00 00 00 00 09 02 27 00 01 01 00 80 32 09 04 00 00 03 ff 00 00 00 07 05 82 03 20 00 01 07 05 01
        02 20 00 00 07 05 81 02 20 00 00 09 02 27 00 01 01 00 80 32 09 04 00 00 03 ff 00 00 00 07 05 82
        03 20 00 04 07 05 01 02 20 00 00 07 05 81 02 20 00 00 12 01 01 01 00 00 00 40 7b 06 03 23 00 03
        01 02 00 01
searching in [/tmp/gadget-fTMRsk]
Starting setup writer thread (1710) for EP00.
Starting setup reader thread (1709) for EP00.
gadgetfs: 1 events received
gadgetfs: event 1
gadgetfs: 1 events received
gadgetfs: event 2
gadgetfs: 1 events received
gadgetfs: event 1
gadgetfs: 1 events received
[80 06 00 03 00 00 ff 00]
LibUSB> 80 06 00 03 00 00 ff 00
LibUSB< 04 03 09 04
[80 06 00 03 00 00 04 00]: 04 03 09 04
gadgetfs: 1 events received
[80 06 02 03 09 04 ff 00]
LibUSB> 80 06 02 03 09 04 ff 00
LibUSB<
        2c 03 55 00 53 00 42 00 2d 00 53 00 65 00 72 00 69 00 61 00 6c 00 20 00 43 00 6f 00 6e 00 74 00
        72 00 6f 00 6c 00 6c 00 65 00 72 00
[80 06 02 03 09 04 2c 00]:
        2c 03 55 00 53 00 42 00 2d 00 53 00 65 00 72 00 69 00 61 00 6c 00 20 00 43 00 6f 00 6e 00 74 00
        72 00 6f 00 6c 00 6c 00 65 00 72 00
gadgetfs: 1 events received
[80 06 00 03 00 00 ff 00]
LibUSB> 80 06 00 03 00 00 ff 00
LibUSB< 04 03 09 04
[80 06 00 03 00 00 04 00]: 04 03 09 04
gadgetfs: 1 events received
[80 06 01 03 09 04 ff 00]
LibUSB> 80 06 01 03 09 04 ff 00
LibUSB<
        32 03 50 00 72 00 6f 00 6c 00 69 00 66 00 69 00 63 00 20 00 54 00 65 00 63 00 68 00 6e 00 6f 00
        6c 00 6f 00 67 00 79 00 20 00 49 00 6e 00 63 00 2e 00
[80 06 01 03 09 04 32 00]:
        32 03 50 00 72 00 6f 00 6c 00 69 00 66 00 69 00 63 00 20 00 54 00 65 00 63 00 68 00 6e 00 6f 00
        6c 00 6f 00 67 00 79 00 20 00 49 00 6e 00 63 00 2e 00
gadgetfs: 1 events received
[80 06 02 03 09 04 ff 00]
LibUSB> 80 06 02 03 09 04 ff 00
LibUSB<
        2c 03 55 00 53 00 42 00 2d 00 53 00 65 00 72 00 69 00 61 00 6c 00 20 00 43 00 6f 00 6e 00 74 00
        72 00 6f 00 6c 00 6c 00 65 00 72 00
[80 06 02 03 09 04 2c 00]:
        2c 03 55 00 53 00 42 00 2d 00 53 00 65 00 72 00 69 00 61 00 6c 00 20 00 43 00 6f 00 6e 00 74 00
        72 00 6f 00 6c 00 6c 00 65 00 72 00
gadgetfs: 1 events received
Sending ACK
[00 09 01 00 00 00 00 00]
LibUSB> 00 09 01 00 00 00 00 00
LibUSB<
Opened EP82
Opened EP01
Opened EP81
Sending ACK
Starting reader thread (1712) for EP81.
Starting writer thread (1717) for EP82.
Starting reader thread (1716) for EP82.
Starting writer thread (1715) for EP01.
Starting reader thread (1714) for EP01.
Starting writer thread (1713) for EP81.
gadgetfs: 1 events received
*retracted bunch of lines of unimportant (i think) communication* 
LibUSB<
Sending ACK
gadgetfs: 1 events received
[21 22 02 00 00 00 00 00]
LibUSB> 21 22 02 00 00 00 00 00
LibUSB<
Sending ACK
gadgetfs: 1 events received
[21 22 00 00 00 00 00 00]
LibUSB> 21 22 00 00 00 00 00 00
LibUSB<
Sending ACK
gadgetfs: 1 events received
[80 06 01 03 09 04 ff 00]
LibUSB> 80 06 01 03 09 04 ff 00
LibUSB<
        32 03 50 00 72 00 6f 00 6c 00 69 00 66 00 69 00 63 00 20 00 54 00 65 00 63 00 68 00 6e 00 6f 00
        6c 00 6f 00 67 00 79 00 20 00 49 00 6e 00 63 00 2e 00
[80 06 01 03 09 04 32 00]:
        32 03 50 00 72 00 6f 00 6c 00 69 00 66 00 69 00 63 00 20 00 54 00 65 00 63 00 68 00 6e 00 6f 00
        6c 00 6f 00 67 00 79 00 20 00 49 00 6e 00 63 00 2e 00

^CReceived SIGINT, stopping relaying...
Finished setup writer thread (1710) for EP00.
Finished writer thread (1713) for EP81.
Finished writer thread (1715) for EP01.
Finished writer thread (1717) for EP82.
Finished setup reader thread (1709) for EP00.
Finished reader thread (1714) for EP01.
^CReceived SIGINT, stopping relaying...
Exiting
^CReceived SIGINT, stopping relaying...
Exiting
^CReceived SIGINT, stopping relaying...  (<--here's me repeatedly hitting Ctrl+C)
Exiting`

In gdb it looks like this
`
[80 06 01 03 09 04 ff 00]
[80 06 01 03 09 04 32 00]:
        32 03 50 00 72 00 6f 00 6c 00 69 00 66 00 69 00 63 00 20 00 54 00 65 00 63 00 68 00 6e 00 6f 00
        6c 00 6f 00 67 00 79 00 20 00 49 00 6e 00 63 00 2e 00
^C
Thread 1 "usb-mitm" received signal SIGINT, Interrupt.
__libc_do_syscall () at ../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:46
46      ../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S: No such file or directory.
(gdb) signal SIGINT
Continuing with signal SIGINT.
Received SIGINT, stopping relaying...
Finished setup reader thread (2118) for EP00.
Finished setup writer thread (2119) for EP00.
Finished writer thread (2122) for EP81.
Finished reader thread (2123) for EP01.
Finished writer thread (2124) for EP01.
Finished writer thread (2126) for EP82.
[Thread 0xb27fa450 (LWP 2126) exited]
[Thread 0xb37fc450 (LWP 2124) exited]
[Thread 0xb3ffd450 (LWP 2123) exited]
[Thread 0xb47fe450 (LWP 2122) exited]
[Thread 0xb5bc1450 (LWP 2119) exited]
[Thread 0xb63c2450 (LWP 2118) exited]
continue
^C
Thread 2 "usb-mitm" received signal SIGINT, Interrupt.
[Switching to Thread 0xb6bc3450 (LWP 2116)]
__libc_do_syscall () at ../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:46
46      in ../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S
(gdb) n
^C
Thread 2 "usb-mitm" received signal SIGINT, Interrupt.
__libc_do_syscall () at ../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:46
46      in ../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S
(gdb) Quit
(gdb) q
`

I don't have any idea where to start with this

RoganDawes, 2020-06-15:
There seems to be a thread waiting for a blocking(?) read to complete, before checking the exit flag. For example, press one more key on the physical keyboard after hitting Ctrl-C in the terminal, and you should see usb-mitm terminate.
I'm no C expert, unfortunately, so have no idea how to fix this.

zapa1928, 2020-06-15:
@RoganDawes Unfortunately another key-presses don't terminate the process.

RoganDawes, 2020-06-15:
Interesting! It does for me.

zapa1928, 2020-08-03:
after adding 
`sigaction(SIGINT, &action, NULL);`
in file usb-mitm.cpp line 69 it does break after third Ctrl+C

But expected behavior would be to stop after hitting it once.
Can someone explain it to me?

MartinLoeper, 2021-01-25:
It looks like Manager.cpp blocks in the following code section in order to wait for the reader threads to join:
```
if (in_readers[i]) {
	if (in_readerThreads[i].joinable()) {
		in_readerThreads[i].join();
	}
	delete(in_readers[i]);
	in_readers[i]=NULL;
}
```

I am no cpp guy either, but I'll try to come up with a solution and post it here.

Leon90PL, 2021-02-10:
DeviceProxy_LibUSB::receive_data calls libusb_bulk_transfer with timeout 0, so timeout is unlimited. Remove
timeout = 0; from void DeviceProxy_LibUSB::receive_data and thread will join.

PR #84 (2021-09-23): Allow usb-mitm to quit on Ctrl+C
Partially resolves #82
closed 2021-09-23 by straithe