usb-tools/USBProxy-legacy #85: Stopped transmission

Leon90PL, 2021-02-08:
Hi,
I connected some printer, usb-mitm started but transmission stopped after few seconds. Could some one take a look and check what's wrong? I added some extra logs. I'm running it on NanoPi NEO

Running under kernel 5.3.0-rc7
CP: String DeviceProxy = DeviceProxy_LibUSB
CP: String HostProxy = HostProxy_GadgetFS
Loading plugins from /usr/local/lib/USBProxy/
vendorId=ffffffff
productId=ffffffff
cleaning up /tmp
removing 1
Made directory /tmp/gadget-oBPfqK for gadget
Printing Config data
	Strings: 2
		DeviceProxy: DeviceProxy_LibUSB
		HostProxy: HostProxy_GadgetFS
	Vectors: 1
		Plugins:
			PacketFilter_StreamLog
Pointer: 1
		PacketFilter_StreamLog::file: 0xb6a2fd10
Connected to device: 0613:2043@02

Error sending setup packet: Pipe error
Device: 12 01 00 02 00 00 00 40 13 06 43 20 01 01 01 02 03 01
  Manufacturer: Printer
  Product:      USB Printer
  Serial:       Printer
	*Config(1): 09 02 37 00 02 01 00 c0 00
		Interface(0):
			*Alt(0): 09 04 00 00 02 ff 00 00 05
			   Name: USB COM
				EP(84): 07 05 84 02 40 00 00
				EP(02): 07 05 02 02 40 00 00
		Interface(1):
			*Alt(0): 09 04 01 00 02 07 01 02 04
			   Name: USB Prn
				EP(84): 07 05 84 02 40 00 00
				EP(02): 07 05 02 02 40 00 00
old bInterval: 00
calculated new bInterval: 80000000
old bInterval: 00
calculated new bInterval: 80000000
old bInterval: 00
calculated new bInterval: 80000000
old bInterval: 00
calculated new bInterval: 80000000

	00 00 00 00 09 02 37 00 02 01 00 c0 00 09 04 00 00 02 ff 00 00 05 07 05 84 02 40 00 00 07 05 02
	02 40 00 00 09 04 01 00 02 07 01 02 04 07 05 84 02 40 00 00 07 05 02 02 40 00 00 09 02 37 00 02
	01 00 c0 00 09 04 00 00 02 ff 00 00 05 07 05 84 02 40 00 00 07 05 02 02 40 00 00 09 04 01 00 02
	07 01 02 04 07 05 84 02 40 00 00 07 05 02 02 40 00 00 12 01 00 02 00 00 00 40 13 06 43 20 01 01
	01 02 03 01
searching in [/tmp/gadget-oBPfqK]
Starting setup writer thread (6082) for EP00.
Starting setup reader thread (6081) for EP00.

event 1
Connect
gadgetfs: event 4
Suspend
gadgetfs: event 2
Disconnect
gadgetfs: event 4
Suspend
gadgetfs: event 1
Connect
gadgetfs: event 2
Disconnect
gadgetfs: event 1
Connect
gadgetfs: event 3
Setup
Request Type 128
Request 6
Index 0
Len 255
[80 06 00 03 00 00 ff 00]
[80 06 00 03 00 00 04 00]: 04 03 09 04
gadgetfs: event 3
Setup
Request Type 128
Request 6
Index 1033
Len 255
[80 06 03 03 09 04 ff 00]
[80 06 03 03 09 04 2c 00]: 
	2c 03 45 00 50 00 49 00 43 00 20 00 34 00 33 00 30 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
	00 00 00 00 00 00 00 00 00 00 00 00
gadgetfs: event 3
Setup
Request Type 128
Request 6
Index 0
Len 255
[80 06 00 03 00 00 ff 00]
[80 06 00 03 00 00 04 00]: 04 03 09 04
gadgetfs: event 3
Setup
Request Type 128
Request 6
Index 1033
Len 255
[80 06 02 03 09 04 ff 00]
[80 06 02 03 09 04 26 00]: 
	26 03 49 00 74 00 68 00 61 00 63 00 61 00 20 00 55 00 53 00 42 00 20 00 50 00 72 00 69 00 6e 00
	74 00 65 00 72 00
gadgetfs: event 3
Setup
Request Type 128
Request 6
Index 0
Len 255
[80 06 00 03 00 00 ff 00]
[80 06 00 03 00 00 04 00]: 04 03 09 04
gadgetfs: event 3
Setup
Request Type 128
Request 6
Index 1033
Len 255
[80 06 02 03 09 04 ff 00]
[80 06 02 03 09 04 26 00]: 
	26 03 49 00 74 00 68 00 61 00 63 00 61 00 20 00 55 00 53 00 42 00 20 00 50 00 72 00 69 00 6e 00
	74 00 65 00 72 00
gadgetfs: event 3
Setup
Request Type 0
Request 9
Index 0
Len 0
Control ACK
Sending ACK 0
[00 09 01 00 00 00 00 00]
Try to open musb-hdrc
Opened EP84
Try to open musb-hdrc
Opened EP02
Try to open musb-hdrc
Fail on open EP84 16 Device or resource busy
Starting writer thread (6087) for EP02.
Starting reader thread (6088) for EP84.
Sending ACK 0
Starting reader thread (6086) for EP02.
Starting writer thread (6089) for EP84.
gadgetfs: event 3
Setup
Request Type 128
Request 6
Index 1033
Len 255
[80 06 05 03 09 04 ff 00]
[80 06 05 03 09 04 10 00]: 10 03 55 00 53 00 42 00 20 00 43 00 4f 00 4d 00
gadgetfs: event 3
Setup
Request Type 128
Request 6
Index 1033
Len 255
[80 06 04 03 09 04 ff 00]
[80 06 04 03 09 04 10 00]: 10 03 55 00 53 00 42 00 20 00 50 00 72 00 6e 00
gadgetfs: event 3
Setup
Request Type 161
Request 0
Index 256
Len 1009
[a1 00 00 00 00 01 f1 03]
[a1 00 00 00 00 01 2e 00]: 
	00 2e 4d 46 47 3a 41 34 32 30 38 43 30 31 30 32 39 35 2d 3b 4d 44 4c 3a 45 34 33 30 20 50 63 4f
	53 3b 43 4c 53 3a 50 52 49 4e 54 45 52 3b

gadgetfs: event 2
Disconnect
gadgetfs: event 1
Connect
gadgetfs: event 2
Disconnect
gadgetfs: event 1
Connect
gadgetfs: event 3
Setup
Request Type 0
Request 9
Index 0
Len 0
Control ACK
Sending ACK 0
[00 09 01 00 00 00 00 00]
Try to open musb-hdrc
Fail on open EP84 16 Device or resource busy
Sending ACK 0
gadgetfs: event 2
Disconnect
gadgetfs: event 1
Connect
gadgetfs: event 2
Disconnect
gadgetfs: event 1
Connect
gadgetfs: event 3
Setup
Request Type 0
Request 9
Index 0
Len 0
Control ACK
Sending ACK 0
[00 09 01 00 00 00 00 00]
Try to open musb-hdrc
Fail on open EP84 16 Device or resource busy
Sending ACK 0
02[3]: 01 5a 0d
02[3]: 01 5a 0d
gadgetfs: event 2
Disconnect
gadgetfs: event 1
Connect
gadgetfs: event 2
Disconnect
gadgetfs: event 1
Connect
gadgetfs: event 3
Setup
Request Type 0
Request 9
Index 0
Len 0
Control ACK
Sending ACK 0
[00 09 01 00 00 00 00 00]
Try to open musb-hdrc
Fail on open EP84 16 Device or resource busy
Sending ACK 0
gadgetfs: event 2
Disconnect
gadgetfs: event 1
Connect
gadgetfs: event 2
Disconnect
gadgetfs: event 1
Connect
gadgetfs: event 3
Setup
Request Type 0
Request 9
Index 0
Len 0
Control ACK
Sending ACK 0
[00 09 01 00 00 00 00 00]
Try to open musb-hdrc
Fail on open EP84 16 Device or resource busy
Sending ACK 0

gadgetfs: event 2
Disconnect
gadgetfs: event 1
Connect
gadgetfs: event 2
Disconnect
gadgetfs: event 1
Connect
gadgetfs: event 3
Setup
Request Type 0
Request 9
Index 0
Len 0
Control ACK
Sending ACK 0
[00 09 01 00 00 00 00 00]
Try to open musb-hdrc
Fail on open EP84 16 Device or resource busy
Sending ACK 0

gadgetfs: event 4
Suspend
gadgetfs: event 2
Disconnect
gadgetfs: event 1
Connect
gadgetfs: event 2
Disconnect
gadgetfs: event 1
Connect
gadgetfs: event 3
Setup
Request Type 128
Request 6
Index 0
Len 255
[80 06 00 03 00 00 ff 00]
[80 06 00 03 00 00 04 00]: 04 03 09 04
gadgetfs: event 3
Setup
Request Type 128
Request 6
Index 1033
Len 255
[80 06 03 03 09 04 ff 00]
[80 06 03 03 09 04 2c 00]: 
	2c 03 45 00 50 00 49 00 43 00 20 00 34 00 33 00 30 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
	00 00 00 00 00 00 00 00 00 00 00 00

END of transmission

mweal-ed, 2021-02-09:
I have not looked at this for about a year, and no longer have access to the full code base I used to fix this in my code, so this is may be kind of incomplete.

usbproxy has trouble with printers that use Zero Length Packets (ZLP) for flow control.  I believe I modified  RelayReader::relay_read() to handle this as below

`
void RelayReader::relay_read() {
	if (!endpoint) {
		relay_read_setup();
		return;
	}

	__u8* buf;
	int length;
	unsigned zlpCount = 0;
	fprintf(stderr,"Starting reader thread (%ld) for EP%02x.\n",gettid(),endpoint);
	while (!_please_stop) {
		buf = nullptr;
		length = 0;

		proxy->receive_data(endpoint,attributes,maxPacketSize,&buf,&length, READ_TIMEOUT_MS);
		if (_please_stop)
		{
			free(buf);
			break;
		}
		if(length)  {
			_sendQueue->enqueue(std::make_shared<Packet>(endpoint, buf, length));
			zlpCount = 0;
		}
		// Receive data can return two different results for a length of
		// zero.  The two are differentiated by the value of buf.  If
		// buf == nullptr there was no data received.  If buf != nullptr
		// a ZLP was received.
		//
		// Zero Lenth Packets (ZLP) can be a problem.  If all ZLPs are
		// passed the performance will suffer and buffers may
		// overflow.  ZLP cannot be completely blocked as they are
		// required in some cases by the usb protocols (i.e. used for
		// bulk transfers that are less than the requested size and a
		// multiple of the maximum packet size).  I have also found that
		// some drivers expect ZLP for no data and will have problems
		// if they just get NAKs (gadgetfs default way to present no
		// data).		
		else if (nullptr != buf) {

			if ((!zlpCount) || _sendQueue->empty() ) {
				_sendQueue->enqueue(std::make_shared<Packet>(endpoint, buf, length));
				++zlpCount;
			} else {
			        ++zlpCount;
				free(buf);
			}
		}
	}
	fprintf(stderr,"Finished reader thread (%ld) for EP%02x.\n",gettid(),endpoint);
	_please_stop = false;
}`

Leon90PL, 2021-02-09:
I modified void RelayReader::relay_read but it looks like it's not my case. New code is never reached.
gadgetfs: event 2
gadgetfs: event 1
gadgetfs: event 2
gadgetfs: event 1
Sending ACK
[00 09 01 00 00 00 00 00]
Fail on open EP84 16 Device or resource busy
Sending ACK
_sendQueue->enqueue len != 0 
02[3]: 01 5a 0d
_sendQueue->enqueue len != 0 
02[3]: 01 5a 0d
gadgetfs: event 2
gadgetfs: event 1
gadgetfs: event 2
gadgetfs: event 1
Sending ACK
[00 09 01 00 00 00 00 00]
Fail on open EP84 16 Device or resource busy
Sending ACK
gadgetfs: event 2
gadgetfs: event 1
gadgetfs: event 2
gadgetfs: event 1
Sending ACK
[00 09 01 00 00 00 00 00]
Fail on open EP84 16 Device or resource busy
Sending ACK
gadgetfs: event 2
gadgetfs: event 1
gadgetfs: event 2
gadgetfs: event 1
Sending ACK
[00 09 01 00 00 00 00 00]
Fail on open EP84 16 Device or resource busy
Sending ACK
gadgetfs: event 4
gadgetfs: event 2
gadgetfs: event 1
gadgetfs: event 2
gadgetfs: event 1
[80 06 00 03 00 00 ff 00]
[80 06 00 03 00 00 04 00]: 04 03 09 04
[80 06 03 03 09 04 ff 00]
[80 06 03 03 09 04 2c 00]: 
	2c 03 45 00 50 00 49 00 43 00 20 00 34 00 33 00 30 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
	00 00 00 00 00 00 00 00 00 00 00 00

Stop transmission

mweal-ed, 2021-02-10:
I took a closer look at your log files, and yes, this will probably not solve your problem (although you may need this fix  after you fix the other problems).   

I do seem to remember there was some sort of problem with the handling of disconnects, but I can't remember how I handled that.  

You also seem to be having some problems before that (Error sending setup packet: Pipe error)... can't be of much help with that.

Leon90PL, 2021-02-11:
I grabbed USB packets from NanoPi using tcpdump. It looks like host doesn't like response for SET_CONFIGURATION, am I right?

No.     Time           Source                Destination           Protocol Info
      1 0.000000       host                  8.2.0                 USB      GET DESCRIPTOR Request DEVICE

Frame 1: 64 bytes on wire (512 bits), 64 bytes captured (512 bits) on interface 0
USB URB
    [Source: host]
    [Destination: 8.2.0]
    URB id: 0x00000000d63ee800
    URB type: URB_SUBMIT ('S')
    URB transfer type: URB_CONTROL (0x02)
    Endpoint: 0x80, Direction: IN
    Device: 2
    URB bus id: 8
    Device setup request: relevant (0)
    Data: not present ('<')
    URB sec: 1612968137
    URB usec: 556648
    URB status: Operation now in progress (-EINPROGRESS) (-115)
    URB length [bytes]: 40
    Data length [bytes]: 0
    [Response in: 2]
    Interval: 0
    Start frame: 0
    Copy of Transfer Flags: 0x00000200, Dir IN
    Number of ISO descriptors: 0
URB setup
    bmRequestType: 0x80
    bRequest: GET DESCRIPTOR (6)
    Descriptor Index: 0x00
    bDescriptorType: 0x01
    Language Id: no language specified (0x0000)
    wLength: 40

No.     Time           Source                Destination           Protocol Info
      2 0.000942       8.2.0                 host                  USB      GET DESCRIPTOR Response DEVICE

Frame 2: 82 bytes on wire (656 bits), 82 bytes captured (656 bits) on interface 0
USB URB
    [Source: 8.2.0]
    [Destination: host]
    URB id: 0x00000000d63ee800
    URB type: URB_COMPLETE ('C')
    URB transfer type: URB_CONTROL (0x02)
    Endpoint: 0x80, Direction: IN
    Device: 2
    URB bus id: 8
    Device setup request: not relevant ('-')
    Data: present (0)
    URB sec: 1612968137
    URB usec: 557590
    URB status: Success (0)
    URB length [bytes]: 18
    Data length [bytes]: 18
    [Request in: 1]
    [Time from request: 0.000942000 seconds]
    Unused Setup Header
    Interval: 0
    Start frame: 0
    Copy of Transfer Flags: 0x00000200, Dir IN
    Number of ISO descriptors: 0
DEVICE DESCRIPTOR
    bLength: 18
    bDescriptorType: 0x01 (DEVICE)
    bcdUSB: 0x0200
    bDeviceClass: Device (0x00)
    bDeviceSubClass: 0
    bDeviceProtocol: 0 (Use class code info from Interface Descriptors)
    bMaxPacketSize0: 64
[... 412 lines cut ...]
    [Destination: host]
    URB id: 0x00000000d115f880
    URB type: URB_COMPLETE ('C')
    URB transfer type: URB_CONTROL (0x02)
    Endpoint: 0x80, Direction: IN
    Device: 2
    URB bus id: 8
    Device setup request: not relevant ('-')
    Data: present (0)
    URB sec: 1612968173
    URB usec: 435574
    URB status: Success (0)
    URB length [bytes]: 44
    Data length [bytes]: 44
    [Request in: 21]
    [Time from request: 0.001360000 seconds]
    Unused Setup Header
    Interval: 0
    Start frame: 0
    Copy of Transfer Flags: 0x00000200, Dir IN
    Number of ISO descriptors: 0
STRING DESCRIPTOR
    bLength: 44
    bDescriptorType: 0x03 (STRING)
    bString: EPIC 430

Leon90PL, 2021-02-15:
tcpdump logs https://imgur.com/a/ARsMJGZ

straithe, 2021-11-13:
Hi @Leon90PL . I'm sorry there hasn't been a response to this issue in a while. Do you still need assistance with this issue?

closed 2021-12-12 by straithe