usb-tools/USBProxy-legacy #86: Please note: Nesto started maintaining this project (with samples)

MartinLoeper, 2021-02-15:
Hi there :wave:

I want to drop a quick note here, that we at Nesto forked this repo and use it in our projects actively.
That is, we also provide binary releases and a HOWTO for the Raspberry Pi 4B.
It would be great to transform this into a maintained project again. 
Every help is much appreciated.

https://github.com/nesto-software/USBProxy

palesius, 2021-02-15:
Great to hear!
I haven't had time to devote to this in ages.
Glad someone else has picked it up.
Great to hear about the Pi4 support too!

MartinLoeper, 2021-02-15:
Great to hear from you @palesius. Thanks for all the good work on this project!   

Just a quick question here: I read that Facedancer is doing something similar in the meantime. Do you happen to know what the differences are and if it is worth the time to dig into it and integrate some ideas from there?

I just quickly glanced over it and noticed that it is written in Python...

palesius, 2021-02-15:
I started working on this originally to inject button presses into video
game controllers (Xbox specifically). So, I did mess around with the face
dancer some and from my recollection I did get it to work in a similar
capacity(viewing and changing usb traffic), but it was much much slower.
That made it totally unsuitable for my needs.

I’m not sure how applicable this would be to your needs but
https://blog.gimx.fr
May be worth checking out. They use an atmega32 with custom firmware to
emulate a usb device and play data through it from the PC. That was a road
I started to go down but the difficulty of debugging on that platform drove
me nuts.

Leon90PL, 2021-02-15:
@MartinLoeper did you already tested your forked repo with some thermal printer? Legacy is not working as expected with my device (device not detected correctly) and I'm kinda stucked. https://imgur.com/a/ARsMJGZ

MartinLoeper, 2021-02-15:
@Leon90PL Yes, our fork works with the Toshiba TRST printer series.

straithe, 2021-11-05:
@MartinLoeper I would be happy to transfer this repository over to you if you are interested.

MartinLoeper, 2021-11-07:
Hi @straithe, I cannot promise that I will actively develop this project. At the beginning of the year, when the nesto-software fork was created, we actively developed and applied this software in the field as part of the [proxy suite](https://github.com/nesto-software/ProxySuite). We needed a quick solution to intercept data from a pos <-> printer connections. We showed that you can extract data from printing protocols without messing with existing customer equipment.

However, the risk of running this stuff purely in software (quasi "in-band") is very high since an outage of our commodity equipment (namely: Raspberry Pi 4B boards) leads to an outage of customer printing equipment. That is why I would really like to see and participate in the future of usb-proxy which IMO is [USBProxy2](https://github.com/nesto-software/USBProxy2) based on the LUNA board.

That being said, I do not want to decline the offer to maintain this project with a more **educational focus** rather than a commercial (i.e. production grade) focus that our GitHub fork kinda has.

straithe, 2021-11-13:
Hi, @MartinLoeper. Myself and the folks at Great Scott Gadgets will be happy to review any pull requests that anyone wants to make to this repository, but we will not be actively developing it ourselves. I myself will also be trying to provide support when I can. Since we are going to be a little bit more active with this repository I will close this issue for now. Thank you for all your hard work!

closed 2021-11-13 by straithe